Ultimaker/Cura #18642: just not slicing, but in v 5.6.0 it does

Jan-Snijder, 2024-03-17:
### Cura Version

5.7.0 b1

### Operating System

windows 11

### Printer

ender 3 pro

### Describe model location

mid bed or everywhere else

### Describe your model

just a simple motorcycle part

### Add your .zip here ⬇️

[CE3PRO_indian-motor2.zip](https://github.com/Ultimaker/Cura/files/14627893/CE3PRO_indian-motor2.zip)

Jan-Snijder, 2024-03-19:
> Hi @Jan-Snijder,
> Thanks for raising the issue and including the file, testing the beta and reporting bugs helps us to develop a more stable product that works for more people. We have tried to slice the file on several computers, and we couldn't get cura to crash. Can you confirm that the exact model shared does not work for you?
> Thanks,
> Casper

Right, this is the last time I share crashes with you.It never is on Cura but always on the user.Goodbye cura.Sent from Android device

insanemal, 2024-03-21:
I have encountered the same issue with 5.7.0 b1

Nothing slices. They slice successfully in 5.6.0.

I am on linux.

What logs or process can I follow to get the information you need now that @Jan-Snijder has stomped off in a rage?

Jan-Snijder, 2024-03-21:
Not in a rage, I have had this multiple times here. Its not only unbelievable but insane that only the users have problems on their machines and the devs don't.Meaning that their computers run some some sw that we don't whitch masks the bugs somehow.Sent from Android device

casperlamboo, 2024-03-21:
> I have encountered the same issue with 5.7.0 b1
>
> Nothing slices. They slice successfully in 5.6.0.
>
> I am on linux.
> 
> What logs or process can I follow to get the information you need now that @Jan-Snijder has stomped off in a rage?

Hi @insanemal , thanks for the report. Strange that _nothing_ slices for you. The most likely reason why it doesn't work is the newly added gradual flow feature which works as a plugin, which at the moment of release didnt quite work on every system. However, this feature has already been introduced back in version 5.5.

Can I ask you to try to disable this feature and see if that helps with the slicing?

[image]

Sharing logs definitely helps us solve the issue. Could you share the following files from your configuration folder?
```
cura.log
CuraEngineGradualFlow.log
``` 
The configuration folder can be located by clicking help -> show configuration folder from within the cura application. It also helps to perform the "fail to slice" action right before sharing the logs. This causes the error logs to be the latest entry in the log files.

Thanks,
Casper

insanemal, 2024-03-21:
I'm also seeing other odd behaviour. When opening or importing STL files, it kinda goes off into the woods. By that I mean, you select your file, hit open and the window closes (or pick something from open recent) and then nothing for a random, but long, period of time. 

Sometimes I have to import/open (or drag a file onto cura) a second file, then both load at once. 

Very odd. 

I actually had gradual flow disabled. I tried slicing with it disabled and it failed then enabled it and tried again. Still failed.

I'll include one of the STL's I was working with in the hopes it helps.

[cura.log](https://github.com/Ultimaker/Cura/files/14693253/cura.log)
[CuraEngineGradualFlow.log](https://github.com/Ultimaker/Cura/files/14693254/CuraEngineGradualFlow.log)
[xbox_stand_cable.zip](https://github.com/Ultimaker/Cura/files/14693292/xbox_stand_cable.zip)

insanemal, 2024-03-21:
Oh also please ignore the MBot3D errors. I'll remove those manually myself. That printer does not exist anymore.

Jan-Snijder, 2024-03-21:
Which prooves my opinion of the devs are missing important stuff when doing updates of whatever and are not running clean instances on clean machines when programming, causing introduction of old bugs over and over again ?Sent from Android device

insanemal, 2024-03-21:
It proves nothing. Regressions happen for multiple reasons. I think it's perfectly reasonable to ask for logs if they can't reproduce it from the model. It's not like they are mind readers or can telepathically read the memory state of your computer over the internet or something. 

I don't understand your rage, it's a bit ridiculous. I  mean "we can't reproduce it here with this model can you get us logs?" "<insert child like temper tantrum>" isn't exactly a sensible way to react to a reasonable request.

It could be your anti-virus, or any number of things. There are probably more combinations of hardware and software for PC's than stars in the sky. 

But no, your right, they should totally just know. 

Plus they owe you for selecting their free software to use. If not for you they would be bankrupt or something because it's people like you, who didn't buy their hardware or pay for their software, that keeps their company afloat.

Jan-Snijder, 2024-03-21:
Then explain to me why v 5.6.0 slices without problems the same stl ?I dare you to.Sent from Android device

I don't understand your rage, it's a bit ridiculous. I  mean "we can't reproduce it here with this model can you get us logs?" "" isn't exactly a sensible way to react to a reasonable request.

insanemal, 2024-03-21:
Because there is a new bug that is probably caused by something in your unique setup?

Like who knows what extra plugins you do or don't have. Who knows what antivirus or firewall settings you have.

Your basically saying "You replaced my motor and now it doesn't work. You've broken my new head gasket!. No you can't diagnose the issue."

insanemal, 2024-03-22:
@casperlamboo 

I just had a thought to check my dmesg output

`[240463.321178] CuraEngine[483241]: segfault at 10 ip 00005a8b08eded8f sp 000077ee50bff140 error 6 in CuraEngine[5a8b08d42000+673000] likely on CPU 9 (core 1, socket 0)
[240463.321194] Code: 1f 40 00 48 8b 73 10 4c 89 ff e8 3c 69 fe ff 48 8b 73 08 4c 89 ef e8 30 69 fe ff 48 8b 73 10 4c 89 f7 e8 c4 66 fe ff 48 8b 1b <48> 83 40 10 01 48 85 db 75 cf 4c 8b 7c 24 10 66 0f ef c0 bf 18 00
[240474.961408] CuraEngine[484517]: segfault at 10 ip 000056705a635d8f sp 00007b9ff5fff140 error 6 likely on CPU 6 (core 6, socket 0)
[240474.961419] Code: 1f 40 00 48 8b 73 10 4c 89 ff e8 3c 69 fe ff 48 8b 73 08 4c 89 ef e8 30 69 fe ff 48 8b 73 10 4c 89 f7 e8 c4 66 fe ff 48 8b 1b <48> 83 40 10 01 48 85 db 75 cf 4c 8b 7c 24 10 66 0f ef c0 bf 18 00
[240512.589543] CuraEngine[484564]: segfault at 10 ip 00005642dc6b9d8f sp 00007016995ff140 error 6 in CuraEngine[5642dc51d000+673000] likely on CPU 9 (core 1, socket 0)
[240512.589557] Code: 1f 40 00 48 8b 73 10 4c 89 ff e8 3c 69 fe ff 48 8b 73 08 4c 89 ef e8 30 69 fe ff 48 8b 73 10 4c 89 f7 e8 c4 66 fe ff 48 8b 1b <48> 83 40 10 01 48 85 db 75 cf 4c 8b 7c 24 10 66 0f ef c0 bf 18 00
[240530.468612] CuraEngine[484619]: segfault at 10 ip 000063f90eb50d8f sp 000071824f5ff140 error 6 in CuraEngine[63f90e9b4000+673000] likely on CPU 7 (core 7, socket 0)
[240530.468626] Code: 1f 40 00 48 8b 73 10 4c 89 ff e8 3c 69 fe ff 48 8b 73 08 4c 89 ef e8 30 69 fe ff 48 8b 73 10 4c 89 f7 e8 c4 66 fe ff 48 8b 1b <48> 83 40 10 01 48 85 db 75 cf 4c 8b 7c 24 10 66 0f ef c0 bf 18 00
[273985.778849] CuraEngine[507453]: segfault at 10 ip 00005e2ba0a83d8f sp 000074398dfff140 error 6 in CuraEngine[5e2ba08e7000+673000] likely on CPU 13 (core 5, socket 0)
[273985.778863] Code: 1f 40 00 48 8b 73 10 4c 89 ff e8 3c 69 fe ff 48 8b 73 08 4c 89 ef e8 30 69 fe ff 48 8b 73 10 4c 89 f7 e8 c4 66 fe ff 48 8b 1b <48> 83 40 10 01 48 85 db 75 cf 4c 8b 7c 24 10 66 0f ef c0 bf 18 00
[274060.381992] CuraEngine[507665]: segfault at 10 ip 0000599ddad85d8f sp 000072b249fff140 error 6 in CuraEngine[599ddabe9000+673000] likely on CPU 14 (core 6, socket 0)
[274060.382006] Code: 1f 40 00 48 8b 73 10 4c 89 ff e8 3c 69 fe ff 48 8b 73 08 4c 89 ef e8 30 69 fe ff 48 8b 73 10 4c 89 f7 e8 c4 66 fe ff 48 8b 1b <48> 83 40 10 01 48 85 db 75 cf 4c 8b 7c 24 10 66 0f ef c0 bf 18 00`

This would explain the slicing issue. I am unsure if this will help, but figured it was relevant

Jan-Snijder, 2024-03-22:
Btw, after trying to slice some other parts its all the same, v5.7.0 b1 will not slice anything while v5.6.0 slices without any problem.

Both version on the same computer, same printer choosen and exactly the same plugins.

For now I will no longer use v5.7.0 b1 or test it.

insanemal, 2024-03-24:
One more update, I enabled gradual flow in 5.6.0 and everything worked fine. So there is definitely something odd going on with 5.7.0b. 

If there is anything else I can test or logs I can get please let me know