Ultimaker/Cura #18644: 5.7.0 Beta 1 Error when using UCP

chipmunk1977, 2024-03-17:
### Cura Version

5.7.0 Beta-1

### Operating System

Linux Ubuntu

### Printer

Crality CR-X

### Reproduction steps

1. I downloaded the sample UCP project from Thingiverse
2. I accepted the choices listed in the project dialogue box.
3. I sliced the model
4. I then cleared the buildplate

### Actual results

The settings that were imported with the model remained in my slicing profile.
For example Buildplate adhesion was originally Skirt but remained as Brim.  Infill was originally Gyroid, but remained as Lightning-
I also restarted Cura, but the changes were still there.

### Expected results

I would expect the settings to revert to my original settings when the model is unloaded.
Leaving the changes unreverted means that the user has to remember to reset anything that this import has altered.  For me this is a serious bug, as the action is not signalled either before or after it is carried out.
The expectation that the user (whether experienced or new) should remember to do this is both unrealistic and unfair.  This will only result in an avalanche of error reports from people who won't know that a previous print has caused this.

By clearing the buildplate or closing Cura, the changed settings must be reset.  The alternative (leaving it as it currently is) will just result is an awful lot of very unhappy users.

### Add your .zip and screenshots here ⬇️

---

MariMakes, 2024-03-18:
HI @chipmunk1977,

Because this behavior was also the case, for the "regular" project files. But we expect that this might become a bigger problem if these universal project files are being used more and more. So I upgraded your Bug to a feature. 

To resolve this we are considering introducing an easier way to undo changed settings after opening a project file and that the changed settings by a Universal Cura Project are only applied temporarily. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-11691

Until there is a solution in place, please save your changes in a project file or profile before you start a new project. The introduced settings settings should be reset if you start a new project. 
[image]

Thanks for the report! 👍

chipmunk1977, 2024-03-18:
Thanks @MariMakes 
I can follow the original thinking, but would just like to reiterate that the big difference is that with a normal project the user has made the changes. In this case someone else has made the changes to the user's profile. 
At the very least, there should be an interim measure which informs the user extremely clearly what is going on.