Ultimaker/Cura #18648: t800

StephAlpaga, 2024-03-17:
### Cura Version

5.7.0 alpha

### Operating System

windows 11

### Printer

Flsun Super Racer

### Reproduction steps

only slice

### Actual results

lenteur et bloque a mi-chemin du slice

### Expected results

bug

### Add your .zip and screenshots here ⬇️

[image]

MariMakes, 2024-03-18:
Hi @StephAlpaga,

Welcome to the UltiMaker Cura Github 🚀 

Several slicing crashes have been resolved with the 5.7 Beta release.

You can download the Cura version here: [https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1)

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

StephAlpaga, 2024-03-18:
Hi Mariska
Great ! it's ok
Thanks

Stephane