Ultimaker/Cura #18649: Fails when Hole Horizontal Expansion is enabled.

astramateria, 2024-03-17:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Prusa i3 MK3S+ w/ .4mm nozzle

### Name abnormal settings

I am using the default profiles with material specific settings (using PLA). I am using the addon Material Settings

### Describe model location

build plate center

### Describe your model

basic Junimo from Stardew Valley

### Add your .zip here ⬇️

[PI3MK3M_junimo_thicker_stems.zip](https://github.com/Ultimaker/Cura/files/14629189/PI3MK3M_junimo_thicker_stems.zip)

GregValiant, 2024-03-18:
Thanks for the report.
That model has a lot of errors in it and it is made up of 11 separate pieces that have been assembled poorly.  The only hole I see is the depression that the stem sits in.  
After repairing the model and merging all the separate pieces into a single model - it sliced with Hole Horizontal Expansion at 0.075 and at 1.5
[image]

I'm going to remove the bug label and close this.  That model has so many problems there is no way to determine if it's a bug or just a bad model.  I hope you understand.

closed 2024-03-18 by GregValiant

astramateria, 2024-03-18:
Oh wow. I never noticed. I usually ignore model errors because they print well like 99.9999% of the time. I guess this is an exception. Thanks for the help.