Ultimaker/Cura #18651: support top on floor

JayZazu, 2024-03-18:
### Cura Version

5.6.0

### Operating System

windows 10

### Printer

flsun sr with klipper

### Reproduction steps

i change support top distance to 0.25

### Actual results

support top distance increases by 0.25
support initial layer wall and interface decreases by just as much
if support brim is active, innermost lines equal to the wall count is affected
those affected lines crash nozzle into print bed despite crash protections

### Expected results

support top distance increases by 0.25

### Add your .zip and screenshots here ⬇️

[egg dispenser piece.zip](https://github.com/Ultimaker/Cura/files/14632463/egg.dispenser.piece.zip)

GregValiant, 2024-03-18:
Thanks for the report.
I can duplicate the problem.  The Z dips to 0.05 which is the difference between the layer height of 0.30 and your "Support Roof Distance" of 0.25.  That should not happen.
The "Support Z Distance" for the roof and floor should always be exact multiples of the layer height.  When they are not; it appears that Cura is incorrectly resolving the Initial Layer Support Thickness to "Layer Height - Support Roof Distance."
When I set the "Support Roof Distance" to 0.29 it results in numbers like in this line:
;TYPE:SUPPORT
G1 F4500 X9.415 Y-44.631 **Z0.01** E39.5778
and then a few lines later:
G0 F12000 X7.551 Y-43.693 **Z0.3**
G1 F4500 X9.624 Y-44.274 **Z0.01** E40.48817

@HellAholic is on the Cura team and may have a comment about this.  I seem to recall this coming up in other reports.

HellAholic, 2024-03-18:
Hey @GregValiant & @JayZazu ,
We've worked on several tickets related to the support z distance, some are included in the latest beta and others we're still working on to get them in for the stable (fingers crossed).
Example: https://github.com/Ultimaker/CuraEngine/pull/2047

HellAholic, 2024-03-22:
Issue was resolved and will be part of the stable release. [Internal build screenshot below]
[image]

Thanks for the report and help :)

closed 2024-03-22 by HellAholic