Ultimaker/Cura #18654: [5.6] Startup optimiser is over enthusiastic

Sophist-UK, 2024-03-18:
### Cura Version

5.6

### Operating System

Windows 10

### Printer

Dagoma DiscoEasy 200

### Reproduction steps

Obvious

### Actual results

1. Custom profiles - machine is loaded, but custom profiles linked to the machine are not listed in Preferences/Profiles.
2. ArcWelder plugin settings - the ArcWelder plugin enable checkbox is shown, but all other ArcWelder settings don't exist.

Disabling Startup Optimisation brings everything back - ergo Startup Optimiser is breaking Cura functionality.

### Expected results

Startup Optimiser should NOT break Cura functionality - it needs to be **VERY** careful only to disable stuff that is genuinely not necessary.

It is (somewhat obviously) better to have a marginally slower startup and a fully functional Cura than for the optimiser to disable something it shouldn't and screw up the Cura functionality,

### Add your .zip and screenshots here ⬇️

n/a

HellAholic, 2024-03-18:
Hey @Sophist-UK , if you've found an issue with a plugin it would be best to discuss it with the plugin developer. In this case you can contact them on their repository for the plugin https://github.com/fieldOfView/Cura-StartOptimiser

closed 2024-03-18 by HellAholic

Sophist-UK, 2024-03-18:
Apologies - I assumed (incorrectly) that this was a bundled plugin. I will log an issue with Aldo @fieldOfView .

However, as previously commented, I suspect that since it appears to disable stuff that needs to be enabled, it will cause Cura to break and (since it is NOT obvious that this is caused by the Startup Optimizer plugin, then you may well get user reports of broken functionality made here - so this issue is still a useful point of reference.