Ultimaker/Cura #18655: [5.6] Rename Custom Profile results in crash

Sophist-UK, 2024-03-18:
### Cura Version

5.6

### Operating System

Windows 10

### Printer

Dagoma DiscoEasy 200

### Reproduction steps

1. Create a custom profile.
2. Rename custom profile.
3. Click on the old profile.

### Actual results

2. Renamed profile appears in itallics (i.e. like a new profile). Old profile name is still shown. Profile is NOT renamed on  the main screen.
3. Cura crashes.

Cura also crashes if instead of 3 you close the settings pop-up and try changing a setting in the main window.

### Expected results

Rename should be successful. And (obviously) Cura should not crash.

### Add your .zip and screenshots here ⬇️

a

HellAholic, 2024-03-18:
https://github.com/Ultimaker/Cura/assets/28710690/5eeb5171-8803-4b8f-b37f-f046431d577e

I tried to reproduce but it does not crash regardless of what I do. Is there a step I missed or do you change a specific setting in the profile?

HellAholic, 2024-03-18:
@GregValiant can you maybe share the configuration folder as a zip please? 
I added 5 custom profiles, and changed them one by one. I also tried the bottom one.
[image]

GregValiant, 2024-03-18:
Here is the 5.6 folder.
I tried to rename "Trailers PETG" and Cura crashed hard.  I will not be testing it again.  Rebuilding all my custom stuff sucked.
[GV 5.6.zip](https://github.com/Ultimaker/Cura/files/14636795/GV.5.6.zip)

HellAholic, 2024-03-18:
Added a ticket to investigate.
For internal reference: CURA-11751

MariMakes, 2024-03-19:
Hi all, 

This does sound a super nasty bug. Losing your configuration folder is not fun.

I used the supplied configuration folder and tried all types of names, settings and activation orders and also haven't been able to reproduce. It's hard for us to resolves if there are no solid steps to reproduce. 

I don't want you to break your configuration folder but any details, on changed settings, profile names or other details to reproduce this issue are very much welcome. 

[image]

Sophist-UK, 2024-03-20:
Here is the Cura settings directory from AFTER the rename corrupted it (with plugin code files deleted to make the file small enough to upload). In the end I had to recreate the entire profile again.
[5.6.zip](https://github.com/Ultimaker/Cura/files/14672443/5.6.zip)

P.S. See also #18654 which may be a contributing factor.

HellAholic, 2024-03-20:
Hey @Sophist-UK , can you also share your current working Cura settings directory?
I can compare the two to find variations, and also see if there is anything that pops up. Might take me the weekend but I rather not leave this bug alone :P

Sophist-UK, 2024-03-20:
I started afresh - so there will be nothing in common by history - only because I recreated all settings.