Ultimaker/Cura #18656: Preview WebCam gone

EmEiBee, 2024-03-18:
### Cura Version

5.7.0.b1

### Operating System

WIN10pro x64

### Printer

I3 like rebuild

### Reproduction steps

Place a item, slice, send to Repetier...

### Actual results

Jump to Monitor-tab. no webcam- view

### Expected results

Jump to Monitor-tab. webcam- view (as in 5.6.0)

### Add your .zip and screenshots here ⬇️

[Abdeckung Gong, b2.zip](https://github.com/Ultimaker/Cura/files/14637814/Abdeckung.Gong.b2.zip)
[image]

MariMakes, 2024-03-19:
Hey @EmEiBee,

Cura doesn't show non-UltiMaker webcams by default.  

Did you use some kind of plugin to have the webcam preview show?
Could it be related to the plug-in not being up to date?

EmEiBee, 2024-03-19:
... sorry, but no ...
Server is a Repetier where the CAM's are connected. That works well in CURA 5.6.0 and before (like OctoPrint also) but don't in 5.7.0b1.
In the meantime, I have the uninstalled 5.7, and I'm back to 5.6 because there were some other problems too.

EmEiBee, 2024-03-27:
Problem solved "accidentally" in connection with another problem. Details can be found at https://github.com/Ultimaker/Cura/issues/18663

closed 2024-03-27 by EmEiBee