Ultimaker/Cura #18662: Brim not generated for all items on the plate

Barrnet, 2024-03-19:
### Cura Version

5.6

### Operating System

Windows 11

### Printer

Flying Bear Ghost 6

### Reproduction steps

Brim settings used:
Brim minimum length: 50mm
Brim Width: 8mm
Brim line count 16
Brim Distance 0.2
Brim replace support: unchecked
Brim Only on Outside: checked
Brim inside avoid margin: 2.5mm
Smart Brim: checked

### Actual results

Some item on the plate don't have the brim. Is visible in every gcode visualizzer

### Expected results

Brim should be generated for all the items on the plate

### Add your .zip and screenshots here ⬇️

unfortunately I don't saved the project because I noticed the issue only after printing the file. I will attach the .gcode file and the Cura profile used.
[profine and gcode.zip](https://github.com/Ultimaker/Cura/files/14646332/profine.and.gcode.zip)
[image]

Barrnet, 2024-03-19:
[pezzi_voron.zip](https://github.com/Ultimaker/Cura/files/14646483/pezzi_voron.zip)
Ok, I tried to slice the same file and saved the project with only the file affected by the problem.

Barrnet, 2024-03-19:
Disabling the support generation (Tree type) fix this problem with the brim correctly generated.
[image]

GregValiant, 2024-03-19:
Thanks for the report.
In the screenshot it sure looks like the models without brims are not resting on the build plate.  Within the gcode of your first post they don't appear on the "initial layer" at all. That's another indicator that they were floating in air during the slice.
Under "Preferences / Configure Cura" one of the options is "Automatically drop models to build plate".  Enable that and bring the models back in.
Another option is to select each model in turn, and check that the "Z" location is "0".
If neither of those fixes it then recreate the situation in Cura, use the "File | Save Project" command, zip the 3mf file that is created, and post the zip folder here.  
Right now I don't see a bug here, but rather a couple of models resting on the build plate while the others are floating in air.

closed 2024-03-19 by GregValiant