Ultimaker/Cura #18663: Slicer error while slice with Brim

EmEiBee, 2024-03-19:
### Cura Version

5.7.0.b1

### Operating System

WIN10pro x64

### Printer

I3 like rebuild

### Reproduction steps

Place part, set brim, slice

### Actual results

Oooops ... Slicing error

### Expected results

Slice as before with 5.6.0

Comment: I go back to 5.6.0 meanwhile

### Add your .zip and screenshots here ⬇️

[image]
[Namensschild Klingel.zip](https://github.com/Ultimaker/Cura/files/14647606/Namensschild.Klingel.zip)

GregValiant, 2024-03-19:
Thanks for the report.
Please provide a project file and the Cura.log file.  I can slice the model just fine in 5.7beta.  Without knowing your settings this cannot be investigated here.

EmEiBee, 2024-03-19:
... to late. I have kill 5.7 and downgrade to 5.6 
But I think the project file is the same due all settings are as before. I have used that on 5.7 (error) and also on 5.6 (went well). So at minimum I can provide that.

[Namensschild_Klingel.zip](https://github.com/Ultimaker/Cura/files/14649722/Namensschild_Klingel.zip)

saumyaj3, 2024-03-19:
@EmEiBee Hey! Can you confirm that the model you added actually crashes on 5.7.0.beta.1. We checked it on two different PC's here with developers. It slices well for us.

saumyaj3, 2024-03-19:
> yes for sure. Try switch on brim and that's it. I tried several models, all of which generated an error when support was activated, regardless of whether support made sense or not... Wait a minute, I have another one that didn't work either, if I remember correctly... [Abdeckung_Gong,_b2.zip](https://github.com/Ultimaker/Cura/files/14653630/Abdeckung_Gong._b2.zip)
I tried this model too, It slices for me. Can you please again try again the beta release https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta. Could it be the alpha that you had?

EmEiBee, 2024-03-19:
No, it was the beta, to be concrete the file "UltiMaker-Cura-5.7.0-beta.1-win64-X64.msi".
I will try it again. But that will take some days due I have to empty my printer queue first.

To be on the correct way better I ask (then sorry):
I have CURA in /programs/CURA as default. Just install the BETA in the same DIR and overwrite the old version? Or uninstall the old 1st?

saumyaj3, 2024-03-20:
Please uninstall and try installing it again. It has some register variables that need clearing for proper path. And is done during uninstalling.

EmEiBee, 2024-03-27:
...and here I am again...
I've now had some time to look into it again.
Uninstalling the old version and installing the new version didn't work.
Then I uninstalled everything, backed up the settings manually (who knows, who knows...) and manually cleaned up the registry. Then run CCleaner again (it also found a few leftovers), restarted the computer and reinstalled CURA.
That ultimately helped. The errors described are now gone and new errors (apart from a few illogical little things) have not appeared yet.
This also takes care of https://github.com/Ultimaker/Cura/issues/18656#issuecomment-2007424256. The stream is now back.

I can't say exactly which register entry was the cause, but the problem was somewhere there...

For future versions, I would suggest modifying the installer so that it cleans up leftovers from the previous version in the registry and recreates the entries after deinstalling the old version while preserving the user settings. You certainly know much better than I do which registry entries are created during installation, which must be retained as user settings and which are not...

So this thing is done so far...

Greetz
Micha

closed 2024-03-27 by EmEiBee

saumyaj3, 2024-03-28:
Hey @EmEiBee ! Thanks for such detailed feedback. I know why that happened. So if you do the msi installer, it is necessary to uninstall it, to clear the paths you just deleted. However, if you do it with the .exe installer it clears the registry when you delete the path leading to it.