Ultimaker/Cura #18667: Add a combined normal and tree support setting.

T9Air, 2024-03-20:
### Is your feature request related to a problem?

On many of my models, I need support on parts that normal supports from the build plate can not support. On those same models, however, there are also large areas that need normal support, and are not properly supported by tree supports.

### Describe the solution you'd like

Therefore, an additional type of support should be added that does normal support in large areas that need support but will also use tree supports in the other areas that need support, such as those that do not touch the build plate.

### Describe alternatives you've considered

I have found that on other slicers, such as Bambu Slicer and Orca Slicer, they do have this additional type of support.

### Affected users and/or printers

This additional type of support can affect many models.

HellAholic, 2024-03-22:
Hey @T9Air ,
Could you share a model/project that highlights the need for this feature? Without proper context, it's hard to judge the need/impact/priority/reason of a certain feature. Thanks in advance.

There is a plugin by 5axes https://marketplace.ultimaker.com/app/cura/plugins/5axes/CustomSupportCylinder that lets you add custom support. Give it a look and see if this one provides the functionality you're looking for.

Cheers!

modrows1, 2024-03-22:
> Hey @T9Air , Could you share a model/project that highlights the need for this feature? Without proper context, it's hard to judge the need/impact/priority/reason of a certain feature. Thanks in [...]

Using CURRENT tree supports and custom support addon ONLY works fully if using support density. Tree's typically have zero density.

Since the introduction of the current (newer) tree supports - the addon and tree supports do not play together properly - With the old tree supports, the TREE and Custom supports would BLEND together if they interacted - Thus you could have HOLLOW tree support and hollow custom supports in the same area - I used this to strenghten tree supports and/or cocoon models

With the newer tree supports if the tree and custom supports intersect, the TREE part of the combination will try to print mid air (when using zero support density). The only work around is using high support density even with tree supports :(

I may add a ticket and screenshots later, since my account is finally working correctly.

It is a hard issue to describe and very few probably took advantage of the old combined support technique. The last time I checked the addon had not been updated for a few years. Perhaps I will try the newest update.

T9Air, 2024-03-25:
I needed to print a **NON-USABLE** gun as a prop. The problem is that the barrel was best supported with normal support, while the trigger needed tree support, since with normal supports, you would need to be on part of the model, and the gun needed to look perfect.

[image]

HellAholic, 2024-03-28:
We have a ticket to discuss with the team.
Internal reference: CURA-11794

T9Air, 2024-03-31:
Hi, am I supposed to be able to use that link?

HellAholic, 2024-04-01:
Hi, that is for internal reference of Cura team. So not something that you would need to access or use.

MariMakes, 2024-04-05:
Hi @T9Air,

We discussed your request with the team and we believe we will not work on this anytime soon and will therefore defer this issue so we can focus on the most requested and needed features and bug fixes from our community and from Ultimaker

We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.  So, thanks again, and please let us know if you run into any other issues 💪 

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

Predator75, 2025-01-20:
> [image] [image] [image] [image] [image]

2025, Cura 5.9 and the problem of tree support is still present :-(

closed 2026-02-04 by github-actions[bot]