Ultimaker/Cura #18674: Phantom unsupported "bottom" layers created mid-structure, increasing "Skin Removal Width" doesn't fix it

ericspivey, 2024-03-20:
### Cura Version

4.13.1

### Operating System

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Set up skull to slice (it is a complicated structure, and won't slice with Cura v5.x, which is why I'm using 4.13)

### Actual results

skull slices
~5 "phantom" bottom layers (not based on apparent features in the .stl) appear at various levels of the sliced structure, none are supported.
Prints fail when they reach the first of these bottom layers, because filiment is air printed all over the structure.

### Expected results

I would expect it to slice without introducing phantom bottom layers mid-print

### Add your .zip and screenshots here ⬇️

[UMS5_skull asby 240214 intact.zip](https://github.com/Ultimaker/Cura/files/14669077/UMS5_skull.asby.240214.intact.zip)
[Phantom bottom layer screen shots.pdf](https://github.com/Ultimaker/Cura/files/14669082/Phantom.bottom.layer.screen.shots.pdf)

ericspivey, 2024-03-20:
Also, as the title suggests, when I increase the "Skin Removal WIdth" parameter up to even 5 mm, the phantom layers are not supressed.

closed 2024-03-21 by GregValiant

GregValiant, 2024-03-21:
4.13.1 works well.  Of the 5.x versions 5.2.2 is the best.  Starting with 5.3 there have been a lot of problems with models not slicing and Windows 11 compatibility.  A lot of work has been done to 5.7 and the beta is out.  So far it is slicing a lot of models that 5.3 to 5.6 would not.
I use the SmartAvionics fork for spiralize and when 5.x is giving me a headache.

ericspivey, 2024-03-21:
Thanks, good to know. I knew that my meshes were accumulating errors, because I had to do a lot of boolean union/difference to make modifications. However none of the software I tried (Blender, Meshlab, Cura) seemed to have a straightforward method for fixing the mesh. Do you know how the Microsoft 3D Builder repair tool works?

GregValiant, 2024-03-21:
I have no idea., but it seems to work well.  The program isn't very intuitive but I use it to alter model files when I don't have the CAD files.
PrusaSlicer has made some sort of deal with AutoDesk for NetFabb.  I don't know how that one works either, but it was the best when it was web based.
In PrusaSlicer, before starting a project or importing a model, use the "File | Repair STL" tool.  Then export the model from Prusa, bring it into Cura, and you can slice it.