Ultimaker/Cura #18675: Slicing failed - unexpected error

BiggumBuddha, 2024-03-20:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

Ender 3 pro

### Reproduction steps

1. Loaded .stl file and ran mesh tools plugin to fix holes
2. Sliced and received unexpected error with no context

### Actual results

Prompted to report bug

### Expected results

File is sliced for printing

### Add your .zip and screenshots here ⬇️

[cura.zip](https://github.com/Ultimaker/Cura/files/14670674/cura.zip)

saumyaj3, 2024-03-21:
Hey @BiggumBuddha! have you tried our new 5.7 beta https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta. We have fixed a bunch of slicing crashes in the version. I tried your model and it slices. 💯

BiggumBuddha, 2024-03-22:
@saumyaj3 I'd love to try it, but the github link for 5.7.0-beta returns me a page not found error.

saumyaj3, 2024-03-22:
Hey I am sorry for giving you the incorrect link : https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1 here you go 🍡

BiggumBuddha, 2024-03-22:
Fantastic. Thank you!