Ultimaker/Cura #18677: Gravity Knife slicing fail

Moeman101, 2024-03-20:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3 S1

### Name abnormal settings

the model does not slice with default files either

### Describe model location

I have to move the blade really far away from the screws for it to slice

### Describe your model

I have printed this once before with the blade closer to the screws.  

### Add your .zip here ⬇️

[CE3S1_blaid_1_Looser_tolerances_.zip](https://github.com/Ultimaker/Cura/files/14672739/CE3S1_blaid_1_Looser_tolerances_.zip)

GregValiant, 2024-03-21:
Thanks for the report.
I didn't have any trouble slicing the project in either 5.6 or 5.7b on my Win10 system.  I don't see any errors in the models.
There have been numerous reports of Cura not performing correctly under Windows 11.

I'll mark this as a duplicate and the Cura team will look.

Vandresc, 2024-03-21:
Hi @Moeman101 , does it also happen in Beta 5.7?

closed 2024-04-05 by github-actions[bot]