Ultimaker/Cura #18680: Slicing fails when "Z-hop when retracting" setting is enabled

yprater, 2024-03-21:
### Cura Version

5.6.0

### Operating System

MacOS Ventura 13.6

### Printer

Neptune Elegoo 4 Plus

### Reproduction steps

1. Upload model
2. Go to the print settings and enable the aforementioned **Z-hop when retracting** setting
3. Hit slice

### Actual results

An error gets thrown, preventing the G-code from compiling. 

`Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.`

### Expected results

The slice should be successful, the g-code should compile and be exportable.

### Add your .zip and screenshots here ⬇️

Video of error - https://github.com/Ultimaker/Cura/assets/85203120/0f117598-ffb0-4789-8c05-d0198ff90340

I can't give you the project file because apparently it's too big (41 MB and the limit is 25).

Log file in PDF format - [cura.pdf](https://github.com/Ultimaker/Cura/files/14706032/cura.pdf)

GregValiant, 2024-03-21:
Thanks for the report.
I haven't had a problem with this myself, and this is the first report of this behavior.
Try and duplicate the problem with a smaller model (or scale that one down) and then try posting a project file up again.
I have special tools for searching text files, but not for PDF.

yprater, 2024-03-21:
@GregValiant

[EN3P_Bulbasaur.3mf.zip](https://github.com/Ultimaker/Cura/files/14707311/EN3P_Bulbasaur.3mf.zip) - I've scaled this down to 50 percent, error persists. 

[curalog1.odt](https://github.com/Ultimaker/Cura/files/14707621/curalog1.odt) - here's an ODT version of the logs. Github would not allow me to import the log file or an RTF version of it.

GregValiant, 2024-03-21:
Use the "File | Save Project" command to create a 3mf project file.
Zip the file.
You can add the Cura.log file to the zip folder before you post it.

yprater, 2024-03-21:
Can you not use the files I have provided? Not understanding. I've provided all the info requested

GregValiant, 2024-03-21:
The 3mf files are corrupt.  The zip file doesn't seem to be a "real" zip file.  I have no idea what and ODT file is.

yprater, 2024-03-21:
> Use the "File | Save Project" command to create a 3mf project file.
> Zip the file.

- This is how I created the 3mf file

yprater, 2024-03-21:
ODT is OpenDocument - Text

yprater, 2024-03-21:
[cura.log](https://github.com/Ultimaker/Cura/files/14712207/cura.log) - log file

[EN3P_charmander.3mf.zip](https://github.com/Ultimaker/Cura/files/14712706/EN3P_charmander.3mf.zip) - Different model, same error. I scaled it down to 50 percent and opened/re-opened in Cura several times to ensure it works.

GregValiant, 2024-03-21:
That will work.
Leave Z-hops enabled and try changing the "Support Z Distance" to 0.2.  That's your layer height.  There is a reported bug there when the Z distance is not an exact multiple of the layer height.  It seems that enabling Z-hops is bringing it out on this model.

@HellAholic and the Cura team will take a look.

This is in 5.7b with Z-hops enabled and all support Z distances at 0.2.
[image]

HellAholic, 2024-03-22:
Hey @GregValiant & @yprater ,
Thanks for the report and the notification. We're currently working on a ticket based on the generated Sentry reports for the crash [CURAENGINE-2F](https://ultimaker-o7.sentry.io/issues/4894861969/?referrer=github_integration)

I'll use your model to also validate the issue resolve ;)
Internal reference: CURA-11742

HellAholic, 2024-03-22:
Well, that was a quick one :)
Issue has been resolved.
Fix will be included in 5.7 Stable.
[image]

Cheers!