Ultimaker/Cura #18682: number of slower layers doesnt apply

iceman2511, 2024-03-22:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

modified ender 3 with modified marlin

### Reproduction steps

import any objects into slicer
set number of slower layers to a number higher than 1
slice the model

### Actual results

in preview, layers above first layer have faster wall speed than infill speed, the infill speed adheres to the slower after first layer rules, walls dont adhere to the slower after first layers rule, prints at normal wall speed.

### Expected results

both infill and wall speed should be at the same speed after first layer when number of slower layers is enabled

### Add your .zip and screenshots here ⬇️

[image]  <---- enabled setting
[image]   <-----disabled setting

[chainsaw man teeth v1.zip](https://github.com/Ultimaker/Cura/files/14715142/chainsaw.man.teeth.v1.zip)

Asterchades, 2024-03-22:
I suspect this is a duplicate of #17210. My understanding is that this has already been fixed for 5.7 (currently in public beta).

If you would prefer to use a release build then versions prior to 5.5 should be unaffected, or you can set the `Top Surface skin Layers` setting to zero (0). Note that most print definitions and profiles don't actually do anything different with the Top Surface Skin so disabling it is likely to have no effect at all on the resulting print quality, and may even expedite certain prints such as those with embossed or debossed features on upper surfaces.

iceman2511, 2024-03-22:
thank you, i will give the new version of cura a try and see if it works.

iceman2511, 2024-03-22:
update: cura version 5.7.0 beta fixes this issue and slows down layers after first layer properly
[image]

closed 2024-03-22 by iceman2511