Ultimaker/Cura #18684: Localhost Error logging in

HarleyGeek89, 2024-03-22:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Creality Ender-3 S1 Pro

### Reproduction steps

I'm trying to login to my account to sync it and it brings me to the page to login, I enter my credentials

### Actual results

It brings me to a page saying 'This page isn't working, localhost didn't send any data.  ERR_EMPTY_RESPONSE'

This is the url [http://localhost:32118/callback?code=woTCcAEYAR3_HPSVbiYgVvtPL4KxenDf&state=e493265486c9f3b233e3c8ffcd808b47efc49ac4be923c6ee9afa6a4fc505c81]

### Expected results

It should log me in

### Add your .zip and screenshots here ⬇️

Picture
[image]

nallath, 2024-03-22:
Can you please provide the log files?

HarleyGeek89, 2024-03-23:
I'm not sure where I would find the log files location?  I looked under my
program files and I don't see anything in Ultimaker for that.  Or do you
mean the Chrome log files?

HellAholic, 2024-03-25:
Hey @HarleyGeek89,
You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log
MacOS: $USER/Library/Application Support/cura/<Cura version>/cura.log
Ubuntu/Linux: $USER/.local/share/cura/<Cura version>/cura.log

Alternatively, if you open Cura > navigate to the Help > Show Configuration Folder, it will open the location for cura.log
[image]

HarleyGeek89, 2024-03-26:
Thanks for the help!

Here is the information
[cura.log](https://github.com/Ultimaker/Cura/files/14765398/cura.log)

HarleyGeek89, 2024-04-16:
Haven't heard from anyone, not sure what to do to log in

HellAholic, 2024-04-20:
Hey @HarleyGeek89 , sorry it took us a while to get back to you. We were busy with the release of Cura 5.7, so I hope you understand.
I went through the logs you sent over but there is nothing of remark there. Since the issue notes that the local host didn't send any data, there is a chance that the transmission was blocked. Couple of this to check:
- Check the windows firewall and permissions
- Run Cura in administrator mode > so that is right click > Run as admin
- Try the new 5.7 version, there were a few improvements there that might resolve the issue.
- Out side of that, the oldest trick of the IT might do the trick "Turning it off and then back on again"

HarleyGeek89, 2024-04-21:
Thanks, I'm going to give them a try now!  Not a problem at all