Ultimaker/Cura #18687: have a setting similar to minimum layer time, but for individual print areas/objects

iceman2511, 2024-03-22:
### Is your feature request related to a problem?

when printing multiple items, the minimum layer time doesn't apply to per object/print area, which is what is intended, but can also cause some issues, for example, you have 10 objects in a shape of a cone, and you print with normal print speed (for me its 135mm/s), at the bottom of the print it looks fine, but as you go near the top of the layers, the area of the print is very small, which makes it very hard to print the peaks without slowing it down, which it doesn't slow down because the minimum layer time applies to per layer.

### Describe the solution you'd like

my idea solves this issue by making the minimum layer time apply to each print area/object, solving the print head moving too fast at very small print areas.
for how to implement this, what id say is to have a pop-down menu, relating to the minimum layer time, and have it select between  minimum layer time per layer, minimum layer time per object, and minimum layer time per print area.

### Describe alternatives you've considered

increase the minimum layer time, but this only works if you're printing the same objects, or at least similar objects with similar print areas at the same layer heigh, but for different objects with vastly different print areas per layer, this setting wont help too much, and you might as well just slow down the print itself.

### Affected users and/or printers

it would help people who print many parts at once, as well as beginners who don't know about reduced print quality when speed is high for small features, and be able to print fast, while also being able to get good print quality for different objects.

### Additional information & file uploads

to help describe what i mean by "print area"
[image]
this print has 2 print areas, but you can also see that it has 2 objects.

[image]
with this image, you can see at the top left, at the last few layers, there are also 2 print areas, but there is only 1 object.

GregValiant, 2024-03-22:
I was thinking that this could be done with a Support Blocker.  It turned out that it can but only when the block is a "Cutting Mesh".  Unfortunately that leaves extra seams and top/bottom layers in the print.
[image]
[image]

Configuring the block as an Infill Mesh eliminates the extra seams but only the "Infill Speed" is affected.
[image]

This has the look of a bug.  I don't know if it's repairable or if it is something that "is the way it has to be" do to other considerations within the code.  This situation seems to have been around for many versions.

iceman2511, 2024-03-22:
you may think that the support blockers aren't working as expected, but after further thought I think they are working as expected.

firstly for the cutting mesh one, what I think its doing is splitting the model into 2 sperate objects, and treating them as separate objects, which gives the name of the "cutting" mesh.

and for the infill mesh, its pretty straight forward, that it only affects the infill of the object, but it still separating the model into 2 models, but it just makes a shell of the model, consisting of only the walls, and then another object can be put inside the shell of the object.

you can see that for both options, you have to split the body into 2 separate objects, and have them use different parameters to get your desired affect, which also means that the creators of Cura currently don't have a way to have different wall speeds and infill speeds for the same object using support blockers, this could also be why the minimum layer time cant currently be implemented, but I would really hope that there would be a way ton fix that.

iceman2511, 2024-12-05:
im hoping writing a message here will notify the cura team of this feature that i want.