Ultimaker/Cura #18693: Select Face to Align to Build Plate - Broken in 5.7.0Beta

modrows1, 2024-03-22:
### Cura Version

5.7.0.Beta

### Operating System

Windows10Pro

### Printer

All

### Reproduction steps

1. Select Model
2. Rotate
3. Select Face to Align to Build Plate
4. Click surface

### Actual results

Incorrect surface aligns - Unable to ever get correct surface to align

### Expected results

Correct surface aligns

### Add your .zip and screenshots here ⬇️

[image]

HellAholic, 2024-03-25:
Hey @modrows1 , Could you share the project file for this issue or some screenshots?

modrows1, 2024-03-25:
> Hey @modrows1 , Could you share the project file for this issue or some screenshots?

I can provide 3mf or screenshot but it is doing it with every file - click on flat surface for alignment - a totally different surface is aligned with the build plate - repeated clicking never aligns the correct surface

I can restart cura and try again - upload files perhaps  - after work
I will also try with a different printer profile

HellAholic, 2024-03-25:
[image]

The issue you're experiencing could be model related, if we can understand why certain model behaves outside of the expected boundaries, then we can pin-point the issue and hopefully address it.
It should not be dependent on the printer or printer profile, the model (stl) should be enough.

Cheers!