Ultimaker/Cura #18694: Tree Supports + Cylindric Custom Support Interference - Partially Broken Since 5.Xmas Trees Released

modrows1, 2024-03-22:
### Cura Version

5.8.0.Beta

### Operating System

Windows10Pro

### Printer

All

### Reproduction steps

1. Enable Tree Supports with zero Support Density
2. Add custom supports using Cylindric Custom Support addon - Any shape or size - That interferes with automatic Tree supoorts
3. Slice

### Actual results

Within the interference zone of the two supports, the automatic tree supports will begin generation in midair - when using ZERO Support Density

There is no issue when using non-zero support density - But zero tree support density is prefered

### Expected results

Prior to 5 Xmas Tree incorporation, the automatic tree and Cylindric Custom Support supports interacted differently when they intersected - while using zero support denstiy.

The tree supports would generate normally from the build plate. then they would be surrounded by (cocooned) and strenghtened by the Cylindric Custom Support shapes.

This was a great feature to made small, tiny and tall tree limbs strong. Additionally it was great for COCOONING parts with very sharp points aimed at the bed surface. Those sharp points are STILL very hard to support and guarantee successful printing.

Prior to 5Xmas Trees, combining Automatic Trees and Cylindric Custom Support shapes was a great way to support complex and sharp parts with minimal (zero) support density - One of the main features of tree supporting.

Additionally, I use to use Cylindric Custom Support to cocoon parts that I intentionally FLOAT above the build plate.

This may not be a frequent complaint, but I truely miss the interaction possible with the OLD trees and Cylindric Custom Support shapes. For now I am FORCED to use Support Density while using Automatic Tree supports. :(

### Add your .zip and screenshots here ⬇️

[image]

[image]

[image]

[image]

[image]

[image]

HellAholic, 2024-03-25:
This seems like an interaction between tree support and a 3rd party plugin. The best path forward would be discussing it with the plugin developer specifically. 
@5axes, could you maybe take a look here?

5axes, 2024-03-25:
The plugin is just a convenient way to add some additional geometries.  No link  between the plugin and the slicing result.

But Thomas Rham is working on this type of issue with the possibility to add "A Cradle tree support".

Better tree support for pointy overhangs: Cradle Generation :https://github.com/Ultimaker/CuraEngine/pull/1880