Ultimaker/Cura #18700: Slicing Error

danielrbunch, 2024-03-22:
### Cura Version

5.6.0

### Operating System

Mac OS Monterey

### Printer

Prusa i3

### Describe your model

Apex Canister Blue

### Add your .zip here ⬇️

[apex-legends-shield-battery-container-multi-material-model_files.zip](https://github.com/Ultimaker/Cura/files/14728830/apex-legends-shield-battery-container-multi-material-model_files.zip)

GregValiant, 2024-03-23:
Thanks for the report.
I am on Windows 10 and Cura 5.6 sliced this with no problems.  That model has some issues though.
The "blue" canister is made up of 13 sub-meshes and at least one part of the assembly is not "water-tight".

I sliced this with -0.1 Horizontal Expansion to make it easier to see the joints.  If your printer is a dual extruder machine then I can understand this.  If it isn't then all those pieces should probably be merged.  Posting a project file would be helpful and may answer some of my questions.
[image]

danielrbunch, 2024-03-23:
Interesting.  It is a single extruder. I grabbed the model of thingiverse.

When I slice a second time, it is happy.  Thought I would report the error though.

danielrbunch, 2024-03-23:
I've had it do that a few times with different pieces of this set of models.

GregValiant, 2024-03-23:
It's a nice model and was developed for a multi-extruder printer.  Something like a Geeetech A10M would be able to print it in three colors.
Since my Ender is also a single extruder printer I'd play with that model a bit.
Open all three models in Cura.
Select All and then rotate them vertical.
Select All and Merge.
Export the model as an STL.
Open that exported file in MS 3D Builder.  It will ask to repair the model which is good.
Split the model so the lid parts are separate.
Save the lid and the container separately.
Open each saved file and "merge" the separate parts.  The lid will become a single solid and the container will be a single solid.
Then you can bring them into Cura and slice each separately or as one long print.

Anyway, that's what I'd do.  Here is a project file with the model adjusted as described above.  The model only really needs support at a small horizontal shelf near the top of the container.

Back to Github stuff - can we close this?  It did seem to have a good chance that it was a model problem.

This is not a "here's the way it must be done" thing.  You can throw the project away and it won't hurt my feelings.
[GV_apex-canister.zip](https://github.com/Ultimaker/Cura/files/14731178/GV_apex-canister.zip)

With the Support Overhang Angle at 63° support will not develop in the threads.
[image]

danielrbunch, 2024-03-23:
Think it's safe to close.  Thanks for the help.   Haven't gotten into fiddling with Models much.

closed 2024-03-23 by danielrbunch