Ultimaker/Cura #18706: adding another class of infill - reinforcing infill

rdgeorgiev, 2024-03-23:
### Is your feature request related to a problem?

Often 3d prints require additional strength in some or all directions. Of course, this can be achieved by special features of the designs, but why not be implemented in the slicer? Imagine a wall shelf attached to a wall - the force of gravity will try to bend the shelf.

### Describe the solution you'd like

The solution? The builders already use it in the concrete structures. They integrate steel bars. The standard infill provides some reinforcing of the 3d prints, but for now, let's call it "volumetric infill". If we add another type of infill - let's call it "reinforcing infill" within the infill volume and layers this will create a hidden grid of invisible inner walls that will make the "wall shelf" much more resistant to the forces of gravity without changing the appearance.

### Describe alternatives you've considered

One typical solution to consider as an alternative is adding special cavities during the design process. These cavities add extra walls to reinforce the object, but in many cases, these extra walls can be added by the slicer as a second grid of more rigid infill, blended with the standard "volumetric infill".

### Affected users and/or printers

This feature will be in favor of both hobby and professional users with all kinds of printers.

GregValiant, 2024-03-24:
Support Blockers configured as "Infill Mesh" can provide this functionality.  Have you tried them?

[image]

rdgeorgiev, 2024-03-25:
Hi Greg and thank you for your input. I tried the infill blockers, but I found this solution cumbersome (maybe I didn't give enough time to master the settings) and not a real solution, but just a workaround, coming with some downsides. My idea is to make it possible to combine two (why not even more) infills with different settings and these infills to blend one into another and not separate them in zones. Here, I photoshopped an example of combining a grid "volumetric" infill with oriented 3 wall lines as "reinforcing" infill. Please note this is not a real slice, but a Photoshop representation of what I envision as one simple example. In this example, it is obvious that the reinforcing infill lines act in one direction only. If we stick to the wall shelf example, the lines' direction should be from the wall towards the free-handing end of the shelf.

[image]

GregValiant, 2024-03-25:
This is not photoshopped.  There are some interesting effects you can create.
[image]

This bracket keeps my printer stand from moving around as it anchors the rear legs to the wall.  The middle feature is a gusset and would be the preferred method to keep gravity at bay.

I bought a bird feeder and the perches for the birds were very small.  I printed perches that simply slid over the metal ones.
My black PLA bird perches lasted one day in the summer Florida sun.  They really needed gussets (although the white PETG prints have held up).

rdgeorgiev, 2024-03-26:
As far as I understand what I see in the first image, you use a blocker to create a few layers of grid infill and on top of these layers, you change the infill to lines. Am I right? If so, this is a feasible solution if you have enough thickness (your object is 35mm thick) to multiply the infill types as layers, but in some cases, you might be limited by the available volume, while still not a good option to fill it entirely with shell layered. If the object or a portion of it is for example 5 mm thick, this might be insufficient space for a layered reinforcement and too much to use of a volume for top/bottom layers only. Blending two types of infill with separated settings is the best solution I can figure out. The blockers describe a partial volume that can be assigned with different settings. Working with different volumetric zones is still a possible workaround, but not a solution in all cases.

GregValiant, 2024-03-26:
I'm not trying to dissuade you here.  It's just something to think about as I watch the sun come up.

A load that sits on a cantilevered bracket will force the bracket downward.  If the bracket happens to be made from a thermo-plastic then Cold Flow Deformation will always be a problem.  That will be true whether it is a solid item from a mold, or a printed item.
You can increase the resistance to it by increasing the inside structure, but to eliminate it you have to use a material that is less susceptible to Cold Flow or add a strut/gusset.
The strength of any item is more often a function of the design rather than of the internal structure.
With the "hinge" area of the left bracket at 100% infill I have just moved the problem to the point where the infill thins out.  If the 100% infill extended throughout the part, the bracket on the right (with the gusset) will still be stronger and by quite a bit.

[image]

The same problem occurs on printed models when there is a screw and nut joint involved.  The constant compression of the screw joint forces the plastic to move and the screw effectively loosens.
The reinforced screw joint on the right will hold up better than the one on the left.  They will both loosen, the left example will just take longer.

[image]

rdgeorgiev, 2024-03-26:
I really enjoy the conversation here, so don't worry about expressing any thoughts you have. I am only sorry that there aren't other participants yet to share their opinions.

Here is a bedside shelf / tray and tablet holder, I designed recently.

[image]

I tried to maintain the shape and its visual appearance as minimalistic and delicate as possible. In order to ensure stability I used some design workarounds, but I still believe that combining different infills would be a much easier and better solution in this particular and in many other cases. I agree that strength is more often a function of the design, but 3D printing as an additive process builds internal structures anyway, so why not use this opportunity as one more tool to increase strength when it is possible to accomplish this with just a simple setting of the slicer.

I am not trying to convince anyone that multi infill is the only or the best solution, but I truly believe that in some cases it can add valuable properties to the prints.