Ultimaker/Cura #18713: Bug découpage erreur inattendu

PacoPlessis, 2024-03-24:
### Cura Version

5.6.0

### Operating System

Window 11

### Printer

Wanhao D12 230

### Describe model location

pivoter de 180° au centre

### Describe your model

J'ai réussi avec la même pièce précedamment

### Add your .zip here ⬇️

[Plaque correction v2 .zip](https://github.com/Ultimaker/Cura/files/14735616/Plaque.correction.v2.zip)

GregValiant, 2024-03-24:
Thanks for the report.
Try the new 5.7beta version.  This "won't slice" bug was worked on (it mostly affected Win11 machines).

I'll mark this as a duplicate.

HellAholic, 2024-03-25:
Hello there!
I would recommend trying the 5.7 beta but I cannot find that printer definition in Cura. If you're using a forked version of Cura, please refer to their repository for reporting issues.
Apart from that, try running Cura without any plugins and see if that resolves your issue.