Ultimaker/Cura #18714: O fatiamento falhou com um erro não esperado. Por favor considere relatar um bug em nosso issue tracker.

3DMMR, 2024-03-24:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

Elego Neptune 4 MAX

### Name abnormal settings

Suporte arvore

### Describe model location

Não

### Describe your model

Sim corta normalmente

### Add your .zip here ⬇️

O problema diagnosticado foi na Aba suporte / Estrutura do suporte / Árvore  
[image]
[image]

GregValiant, 2024-03-24:
Thanks for the report.
Try the new 5.7beta version.  It seems that many of these "won't slice" bugs were addressed in the new version.

I'll mark this as a duplicate.

3DMMR, 2024-03-24:
Problema resolvido na versão 5.7 beta.
Obrigado!

closed 2024-03-24 by 3DMMR