Ultimaker/Cura #18715: Ender 3V3 KE profile

illudedPerception, 2024-03-24:
### Is your feature request related to a problem?

the current profile i use for ender 3 V3KE is the SE one but i get an error 2252 or something if pressed ok it prints but yeah

### Describe the solution you'd like

A profile for the printer if possible

### Describe alternatives you've considered

the creality program i dont want to even look at no gcode modding, no plugins so a hard no

### Affected users and/or printers

Ender 3 V3KE

GregValiant, 2024-03-24:
The definition files must come from a member of the community or from the manufacturer.
Have you contacted Creality about a Cura definition file for your printer?  Reddit?
If you install something like an Ender 3, does that error still come up?

illudedPerception, 2024-03-25:
> The definition files must come from a member of the community or from the manufacturer. Have you contacted Creality about a Cura definition file for your printer? Reddit? If you install something like an Ender 3, does that error still come up?

Google sent to to this repositorys pr page with some one wanting to merge his KE profile gonna get it tomorrow

closed 2024-03-25 by illudedPerception