Ultimaker/Cura #18718: Tree support fails slicing

V-Roger, 2024-03-25:
### Cura Version

5.6.0

### Operating System

Windows 11 Pro

### Printer

Creality CR-10

### Name abnormal settings

Tree support / build-plate only causes the crash

### Describe model location

no

### Describe your model

Basic quarter-sphere shell

### Add your .zip here ⬇️

[cura_crash_model.zip](https://github.com/Ultimaker/Cura/files/14740415/cura_crash_model.zip)

HellAholic, 2024-03-25:
Hey @V-Roger,
Could you check and see if you're still getting a crash in the 5.7 Beta? We've made quite a few improvements and significantly reduced the number of slice fails.

Cheers!

V-Roger, 2024-03-25:
Sure thing, I'll download it right away.

V-Roger, 2024-03-25:
That's a big yes from me!
Works as expected in 5.7beta, well done :)

[image]

closed 2024-03-25 by V-Roger