Ultimaker/Cura #18719: 100% infill breaks bridging

phazei, 2024-03-25:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Custom

### Reproduction steps

I changed infill to 100% where bridging happened

### Actual results

The bridging went away and it only had regular infill

### Expected results

The bridges stay where they are

### Add your .zip and screenshots here ⬇️

99% infill:
[image]

100% infill:
[image]

101% infill:
[image]

phazei, 2024-03-25:
Also, not directly related, but why are those bridge lines so thin?  Thinner lines are even more difficult to bridge, can't they be regular width?

phazei, 2024-03-25:
[base-side-1.stl.zip](https://github.com/Ultimaker/Cura/files/14747434/base-side-1.stl.zip)

The stl in case that helps

GregValiant, 2024-03-25:
Thanks for the report.

Infill is treated differently when it is at 100% density since 100% plastic doesn't care what pattern is used.  What you are seeing might be a side-effect of that.
For now, I would suggest setting the Infill Density to 99.99% (if that's what you require).

The Cura team will take a look.

HellAholic, 2024-04-29:
Added a ticket.
Internal reference: CURA-11883