Ultimaker/Cura #18721: The extruder selection is not set to start the print

Robert-AUL, 2024-03-25:
### Cura Version

5.7.0 Beta

### Operating System

Windows 11 professional

### Printer

Tenlog TD3

### Reproduction steps

I'm using the two extruders of my printer and the Gcode doesn't include the selection of the first extruder, in this case T1, before starting printing.
You will have the full configuration in the file "TBL_870-Bi_Piege-Frelon-Trou-Conf370g.3mf"

### Actual results

The print start with T0

### Expected results

Start the print with T1

### Add your .zip and screenshots here ⬇️

The complete configuration is in the file "TBL_870-Bi_Piege-Frelon-Trou-Conf370g.zip"
[TBL_870-Bi_Piege-Frelon-Trou-Conf370g.zip](https://github.com/Ultimaker/Cura/files/14747833/TBL_870-Bi_Piege-Frelon-Trou-Conf370g.zip)

GregValiant, 2024-03-25:
Thanks for the report.
As you can see here, Cura correctly starts the print with T1...
; Tenlog TD3 Bi-Ext AVEC BLTOUCH Debut Code demarrage travaille
G28 ; Origine sur tous les axes pour les deux extrudeurs
**T1** ; début initialisation extrudeur T1

At the end of your startup gcode you explicitly set the printer to T0
**T0** ; Sélection de l'extrudeur T0 pour commencer
; Afficher l'impression démarre sur LCD screen
M117 Debut impression ...
; Tenlog TD3 Bi-Ext AVEC BLTOUCH Fin Code demarrage travaille
;
You need to revert to the initial extruder at the end of your startup gcode.
T{initial_extruder_nr}

I'll remove the bug label and close this as it is a problem with the start up gcode and Ultimaker is not responsible for that.  I hope you understand.

closed 2024-03-25 by GregValiant

Robert-AUL, 2024-03-26:
Hello Greg

Thank a lot for your support.

Regards

 

Robert AUREL

Tel +33 778 656 806

Email : ***@***.*** ***@***.***>