Ultimaker/Cura #18722: Clearing the build plate (control-d) doesn't clear the file name so all saved files have the same name.

twostage, 2024-03-25:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender V2

### Reproduction steps

Start Cura, load a file (lets call it FileA.stl), slice and save it.

Clear the build plate, load a different file (lets call it FileB.stl), slice and save it,

### Actual results

FileB will be saved as FileA.gcode not FileB.gcode as the filename persists after clearing the build plate.

### Expected results

Clearing the build plate should clear the file save name so that the saved gcode file reflects the name of the file you load after clearing the build plate. FileB.stl should save as FileB.gcode, not FileA.gcode

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-03-25:
Thanks for the report.
This may be a duplicate of #11276 and there are other reports as well.  The team has this on the backlog.  I wish they would put it in the front row.

@HellAholic have you happened to take a look at this problem?  It's truly annoying when the project name reverts to a previous name (with no alert to the user other than the silent title bar) and a previous gcode file gets over-written by the new one.

HellAholic, 2024-03-25:
Hey @GregValiant,
I'll take a look tomorrow.

GregValiant, 2024-03-25:
Thanks.
I'll leave all the bumper stickers on until I hear back.

twostage, 2024-03-26:
> Thanks for the report. This may be a duplicate of #11276 and there are other reports as well. The team has this on the backlog. I wish they would put it in the front row.
>
> @HellAholic have you [...]

Thanks for this. I did a search first but #11276 was phrased differently, maybe I should have searched more.

I might go back to 5.3 (which worked OK in this respect) until it is resolved.

HellAholic, 2024-03-29:
Hey @GregValiant , we added a fix for the original issue  https://github.com/Ultimaker/Cura/issues/11276 reported and that will be part of the stable (coming to you soon :P)
Unfortunately I had no luck reproducing the issue reported here.
So could you try it out after the Stable release and let me know if this issue is also resolved?

GregValiant, 2024-03-29:
Hi @HellAholic 
I'll wait on the stable and double-check.  I can confirm that 5.7beta has the "changing project name" problem described in #11276 .
To recreate the 11276 problem in 5.7beta:
Open any file.
Save as project.
[image]

Clear build plate.
Import a different model.
[image]

Slice.
Immediately after saving the gcode the project name reverts to the earlier saved project name.  Any gcodes saved after that will have the previous project name.  Unless the user is aware of it - a previous gcode will be over-written.
[image]

HellAholic, 2024-03-29:
@GregValiant, (mind the the interface language :P)
https://github.com/Ultimaker/Cura/assets/28710690/a6ae762f-98d1-4444-bbf8-ec6b1099df1d

GregValiant, 2024-03-29:
That looks good.
(Stupid bug got me again this morning.)

@twostage The original bug seems to be fixed for 5.7 stable.  This report seems to be the same thing.  I'll leave both of them open until 5.7.0 is released.

HellAholic, 2024-04-29:
Hey @GregValiant , have you been bitten by this bug after the 5.7?

GregValiant, 2024-04-29:
I just checked with 5.7.1 and it didn't happen.  This looks fixed to me!