Ultimaker/Cura #18726: I CANT PUT MULTIPLE STLS IN ONE FILE TO PRINT ALL AT ONCE

Lintypine, 2024-03-26:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

Elegoo neptune 3 plus

### Reproduction steps

tried to drag n drop files and then when that failed clicked the file button and couldn't put them in

### Actual results

nothing

### Expected results

have multiple files in one

### Add your .zip and screenshots here ⬇️

[hidden blade (2023 remix) READ DESCRIPTION TUTORIAL POSTED - 5717722.zip](https://github.com/Ultimaker/Cura/files/14753354/hidden.blade.2023.remix.READ.DESCRIPTION.TUTORIAL.POSTED.-.5717722.zip)

GregValiant, 2024-03-26:
Thanks for the report.
I've never had this problem and it wasn't a problem with your files.  Both dragging them in and opening multiple at once worked for me (as it always has).
This is in 5.5.0 (Windows 10) as I don't have 5.4 installed.
[image]

I'll leave this open.  You will want to install [5.7beta](https://github.com/Ultimaker/Cura/releases) and see if it is still an issue.  Several slicing problems and some incompatibility with Win11 have been addressed in 5.7.  Let us know if that works for you.

Lintypine, 2024-03-27:
how do i install 5.7beta?

Lintypine, 2024-03-27:
nvm i didnt scroll all the way down

Lintypine, 2024-03-27:
im so stupid i wasnt doing it right but thank you for alerting me to a newer better version of cura

closed 2024-03-27 by Lintypine