Ultimaker/Cura #18730: why lines not linear?

cihankosar, 2024-03-26:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

ultimaker 2

### Describe your model

why lines not linear?

### Add your .zip here ⬇️

why lines not linear?
[image]

GregValiant, 2024-03-26:
I have no idea.
We ask for project files for a reason.

cihankosar, 2024-03-26:
I tried 3.6.0, 5.6.0 and 5.7.0 versions. Every version has like this problem. I shared all files with cura profile.

[Files.zip](https://github.com/Ultimaker/Cura/files/14758542/Files.zip)

GregValiant, 2024-03-26:
This is in Cura 4.13.1 and I don't see any oddities in the lines.
[image]

This is 5.7beta at layer 291.  It is a single wall print up to this point and there are no oddities in the walls.
[image]

2 layers later the transition from a single extrusion wall to a double extrusion wall is complete.  You can see how much narrower the line widths are.
[image]

So what you are seeing in the Cura versions from 5.0 up is the effect of the Arachne Engine calculating the necessary line widths.  It is mostly a visual thing in the preview but there is often a visible transition in the print.  If you have a machine capable of measuring the variance at that point my guess is that it would be around .01mm to .02mm.

Look at the difference in wall thickness here.  It is why one side (two narrow extrusions) probably looks a little different from the other side (two wide extrusions).
[image]

At layer 313 the left side is still wide but the right side is down to two narrow extrusions.
[image]

I think a lot of this has to do with the way the inside of that model angles up towards the opening.  It appears to be smooth but the constantly changing width of the model means that the slicer has to constantly change the extrusion width.

So is this a bug?  I have no idea.  For the 5.x versions it's the way it works.

This is in PrusaSlicer and I see the same effect.
[image]

The Cura team will take a look.

rburema, 2024-03-27:
Like @GregValiant says, the upper part of the model becomes thicker, so that's part of the explanation.

I think the problem is also that the walls are in thickness very close to the 'Minimum Wall Line Width' (going over and under that value repeatedly). I had better luck with this particular model setting it to 0.38 (rather than 0.34). (The (other) 'Wall Transition ____' parameters might also help, but those really are expert settings...)

cihankosar, 2024-03-28:
Thank you for all answer. Today I tried to modify cad model. I fixed thickness to 0.8 mm in everywhere. Now it is better than before.