Ultimaker/Cura #18733: bed temperature sets to highest temperature frommaterials selectedeven if that material is unused

shrpshtr325, 2024-03-26:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Proforge4 with Klipper

### Reproduction steps

1. load different materials (with different bed temperatures) into all 4 tools (example, t1 ABS t2 PLA t3 PVA t4 PA6)
2. load file to slice
3. select any tool except the tool with the highest bed tmperature (example select PLA on tool 2 for the part, but you have nylon loaded in tool 4)
4. slice file
5. check gcode, bed temperature will be 110C instead of the expected 60C

### Actual results

Bed is set to 110C 

### Expected results

bed is set to 60C since you are only printing with PLA

### Add your .zip and screenshots here ⬇️

[troubleshoot.zip](https://github.com/Ultimaker/Cura/files/14760758/troubleshoot.zip)
[image]

GregValiant, 2024-03-26:
Thanks for the report.
There is only one bed and the bed temperature will be the highest temperature of the "Enabled" extruders.
If you are only using 1 extruder you need to disable the other three.  The bed temperature will then be whatever the remaining extruder is set to.

This is expected behavior so I'll remove the bug label and close this.  I hope you understand.

closed 2024-03-26 by GregValiant

shrpshtr325, 2024-03-26:
While there is clearly one bed, there is an option to print with multiple
extruders where you whould still want the lower bed temperature, such as
using differing materials for supports, the materials that are printed on
the bed should control the bed temperature NOT the highest material loaded
into the machine. There is also no quick easily accessible method to
'disable' unused extruders, it is only accessible through a series of
dropdown menus.