Ultimaker/Cura #18735: Cura 5.7 beta refuses to slice a file at 0.4mm layer height

Motorace1, 2024-03-26:
### Cura Version

5.7 beta

### Operating System

Windows 11

### Printer

Creality Ender 6, on Insanity Automation firmware

### Name abnormal settings

0.4mm layer height, using 0.8mm nozzle. It slices using 0,2mm layer height.

### Describe model location

No difference moving it, scaling it very small it might slice, but not all the time.

### Describe your model

Yes, sliced and printed on this printer before, but at 0.2mm layer height.

### Add your .zip here ⬇️

[16V Head parts tray.zip](https://github.com/Ultimaker/Cura/files/14763553/16V.Head.parts.tray.zip)

GregValiant, 2024-03-26:
Thanks for the report.
This sliced for me on my Windows 10 laptop at both 0.2 and 0.4 layer height, 0.8 nozzle at 0.8 line width.
There have been a lot of reports regarding compatibility problems with Cura and Windows 11.

I'll mark this as a duplicate.  @HellAholic and the Cura team will take a look.

rburema, 2024-03-27:
@Motorace1 -- Can you maybe save this as a normal Project file, rather than an UCP? UCP is for sharing between different kinds of printers and only saves a select few settings.

HellAholic, 2024-03-27:
[image]
It slices without issues in windows 11. If you have made a custom profile for your printer, share a normal project file rather than a UCP, since the UCP does not include all the 98 overrides you have on your profile.

Motorace1, 2024-03-29:
This should be a proper Cura file.
[CE6_16V Head parts tray.zip](https://github.com/Ultimaker/Cura/files/14804866/CE6_16V.Head.parts.tray.zip)

HellAholic, 2024-03-29:
Can reproduce the issue.
Thanks for the report and sharing the file. 👍 
[image]

Motorace1, 2024-03-29:
@HellAholic  I can also add one thing I noticed that might be linked to the issue. If one uses that profile, but at 0.32mm layers (which slices, for some reason) but choose the "concentric bottom layer" instead of lines or zig-zag, it prints the brim and maybe 20% of the bottom layer, then it just stops dead.  I have tried several SD cards, so it is not a card issue. I guess it just did not calculate the whole g-code, even if the preview seems normal.

HellAholic, 2024-03-29:
We need to do a bit of investigation first, I just loaded the model in and pressed slice. Thanks for the additional info :)

HellAholic, 2024-04-02:
Small update, I've added a ticket to the backlog, 
For internal reference CURA-11802.
We resolved a similar issue CURA-11742 in 5.7 Beta, but this seems to be an edge case to that.

HellAholic, 2024-04-02:
Hey @Motorace1 ,
while we're working on a fix, I have a workaround for you. It appears that the combination of your settings/profile with the gradual flow plugin causes the crash.
If you disable the Gradual flow in the settings, then your model slices.
[image]
Probably the setting should be disabled by default but since you're using a custom setting you're getting that issue.
[image]