Ultimaker/Cura #18740: [5.7.0-beta.1] cant slice model

SkyeCypher, 2024-03-27:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Neptune 4 Pro

### Describe model location

i have tried multiple angles buy keep getting the error

### Describe your model

yes model is watertight

### Add your .zip here ⬇️

[left_wing.zip](https://github.com/Ultimaker/Cura/files/14771281/left_wing.zip)

HellAholic, 2024-03-27:
Hey @SkyeCypher ,
Could you please include a project file instead of an stl?
I can reproduce a scenario with tree support enabled that has been resolved in the fixes after the Beta and will be part of the Stable release. However, without a project file I have no idea whether the scenario that I imagined is the same as the one you're experiencing.

SkyeCypher, 2024-03-27:
[Uploading CFFFP_no-supports-wing-left.zip…]()
here you go hope this helps

SkyeCypher, 2024-03-27:
[Uploading CFFFP_no-supports-wing-left.zip…]()

SkyeCypher, 2024-03-27:
[CFFFP_no-supports-wing-left.zip](https://github.com/Ultimaker/Cura/files/14776501/CFFFP_no-supports-wing-left.zip)

HellAholic, 2024-03-27:
Thanks, I can reproduce the issue you're experiencing.
[image]

HellAholic, 2024-03-27:
This one is also fixed :)
Fix will be included in the Stable release.
Thanks for the report and file 👍 

Internal reference: CURA-11742
[image]