Ultimaker/Cura #18742: Supportgenerating Issue when change to extruder 2 for support interface

Taigunata, 2024-03-27:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

Changed Support Interface Extruder to Extruder 2

### Actual results

No support in overhang area anymore

### Expected results

Still support in overhang area

### Add your .zip and screenshots here ⬇️

[Supporttest.zip](https://github.com/Ultimaker/Cura/files/14771824/Supporttest.zip)
[image]
[image]

HellAholic, 2024-03-27:
Hey @Taigunata ,
This is an issue with your settings. Firstly, a **WARNING** that what you're attempting to do is not tested nor approved.

[image]
If you're accepting the risks involved, the reason why you're not seeing the tree support being generated is the support z distance.
PVA is used as Normal support with Z distance of 0 for the interface layer, if that value is changed to a multiplicative of the layer height (for 0.15mm layer height -> value is 0.15 or 0.3 or 0.45,...) then you will see the branches being generated.

closed 2024-03-27 by HellAholic

Taigunata, 2024-03-27:
Thank you for the response and the workaround. But now I am curious why with the standard Support 0 is working but not with the Tree Support, like having a gap between support and print is nessesary.

HellAholic, 2024-03-27:
Short version is that it is not necessary, but it was the fastest way to show the issue is related to the settings.
You can try the 0.2 layer height and 0 support top distance also generates the support.