Ultimaker/Cura #18746: FreeCAD trackpad controls: Dragging seems laggy

oliof, 2024-03-27:
### Cura Version

5.7.0-beta1

### Operating System

POP!_OS Linux 20.04

### Printer

Any

### Reproduction steps

1. Turn on FreeCAD touchpad style navigation
2. Try Dragging the view around  holding down the shift key

### Actual results

Dragging seems to follow the mouse at maybe half speed

### Expected results

Dragging should be at a more natural speed

### Add your .zip and screenshots here ⬇️

Here is a screen cast video of the issue. As you can see, dragging is slow while rotating the view is tracking the mouse movement more closely.
[Screencast from 03-27-2024 05:19:20 PM.webm](https://github.com/Ultimaker/Cura/assets/1230627/8963d8f6-72c6-4c2c-b51e-1070db3ea895)

rburema, 2024-03-28:
Hm, doesn't reproduce when I'm using windows. I'll ask one of my linux-using colleagues to take a look at it.

oliof, 2024-03-28:
I also see that zoom and rotate are fixed on a point and do not center on the mouse pointer as in FreeCAD. Should I file an additional issue for that?

rburema, 2024-03-28:
Normally, yes, but I'll just add it to the points to be discussed for now -- we'll get back on if we'd like an additional issue for it.

(There is a separate preference in Cura for 'zoom towards mouse pointer' though. That'd only solve half of it though, if it's even what you want...)

oliof, 2024-03-28:
"Zoom Towards Mouse Pointer" makes zooming with FreeCAD Trackpad selected work as expected; I would suggest to enable it by default when people choose FreeCAD Trackpad navigation to better match expectations.

oliof, 2024-03-28:
Also, one more comment: This is ona HiDPI screen, which may be a factor.

nallath, 2024-03-29:
I think it might be the HiDPI screen. We couldn't reproduce this on linux & windows with a "normal" screen.

oliof, 2024-03-29:
I will try to switch the display to Low DPI and report back.

oliof, 2024-03-29:
Alright, I did some more testing, and figured out the following:
a) it's resolution dependent. Switching to 100% scaling on a 2256x1504 display amplifies the issue.
b) it's not a problem with the new FreeCAD Touchpad mode, I have the same problem with Cura 5.6 and 5.7 in Cura's native navigation mode.
c) I tested Cura 5.5 and 5.4 for posterity, and they have the issue as well. So this either is an old issue, or an issue with my OS/X11 setup that crept in some time between.