Ultimaker/Cura #18747: have different combing settings for initial layer

iceman2511, 2024-03-27:
### Is your feature request related to a problem?

when combing first layer, it sometimes grabs already printed parts and unadheres it from the build plate, I'm using G10 build plate and is properly cleaned with both soapy water+ isopropyl alcohol, adhesion is completely fine when printing normally, and only has problems when the combing takes place

### Describe the solution you'd like

have a selection box like the normal combing, but for initial layer

### Describe alternatives you've considered

just disable combing, but then you'd have more stringing from filaments like PETG and LW PLA

### Affected users and/or printers

people who use combing, it might not affect users who are using certain build plates, such as buildtak because of its natural adhesive properties without heat, but there are many downsides of using it.

rburema, 2024-03-28:
I've put this on the list of things to discuss with the team.

iceman2511, 2024-12-05:
> I've put this on the list of things to discuss with the team. (Internal ref: [CURA-11791](https://ultimaker.atlassian.net/browse/CURA-11791))

have you discussed about this idea yet, or any progress?

rburema, 2024-12-05:
I see we've let the material & print-processing team know. Maybe @pkuiper-ultimaker knows if this is on the radar?

pkuiper-ultimaker, 2024-12-06:
It is definitly an intresting idea. If you disable combing it will do a retracted move, which should reduce oozing and stringing. So it do not fully agree that the described alternative does not work. That alternative will cause the print to take much longer if you have large retaction distances, which is typically the case for materials that ooze a lot.

More retracted moves can also create blobs on the first layer that the print head can run into also resulting in adhesion failures. A retraction with z hop can also pull the layer loose from the buildplate. However, the overall failure chance might be lower. 

Perhaps (partial) retract/unretraction (+ z hop) during the unretracted combing moves can reduce the ooze over the bottom layer and reduce the chance of hitting something (Internal reference: CURA-4441).

But all the above is just speculation and we just need to test what works. @rburema If it is a very small change to the code I would not investigate deeply and simply implement it and expose it as an advanced setting for the profile makers.