Ultimaker/Cura #18752: The supports are wrong in FLSun v400

AlbertLiv, 2024-03-28:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

FLSun v400

### Name abnormal settings

Only when supports generates it fails

### Describe model location

a little few times with a strange arragment wotks, but it isn´t funtional

### Describe your model

and ender 3 s1 upgrade with a 5015 fan

### Add your .zip here ⬇️

[Ender 3 S1 Duct_5015_Impr3Decible 2.zip](https://github.com/Ultimaker/Cura/files/14784797/Ender.3.S1.Duct_5015_Impr3Decible.2.zip)
Only when i put the supports it fall

rburema, 2024-03-28:
You attached the same file as for the last issue, was this intentional?

rburema, 2024-03-28:
(Also assuming this is about a slice failure, not the supports itself being wrong after slicing.) If so, see my previous reply. As far as I can see, it slices properly in the 5.7.0-beta at least for me.