Ultimaker/Cura #18755: Problemas en segmentacion

pepelud3s, 2024-03-28:
### Cura Version

5.6.0

### Operating System

macos Monterey 12.7.4

### Printer

Ender 3 v2 con marlin

### Reproduction steps

puse una pieza y la aumente , poniendola al 150%

### Actual results

al empezar la segmentacion da un error en el corte

### Expected results

No me segmenta la pieza 

### Add your .zip and screenshots here ⬇️

[image]

HellAholic, 2024-03-29:
Can you try [the 5.7.0-beta](https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1)? We've fixed a lot of the slicing errors we saw in the 5.x-series in the upcoming 5.7.0 version.

pepelud3s, 2024-03-29:
la acabo de probar y me sigue dando el mismo error