Ultimaker/Cura #18761: Loaded Cura project not respecting layout when using spoontabs

gh13, 2024-03-29:
### Cura Version

5.7.0 Beta 1

### Operating System

Windows 10 

### Printer

Ender 3 S1 Pro

### Reproduction steps

Created a project using model & spoontabs
Saved & reloaded project 

### Actual results

Objects apart from one spoontab positioned above the build plate

### Expected results

Objects in the project should respect original positions 

### Add your .zip and screenshots here ⬇️

[Desiccant Box - Box [XL] [HatchBox PLA (Standard), LW-0.4mm, LH-0.2mm, I-25%, T-E.215C, B.65C].zip](https://github.com/Ultimaker/Cura/files/14802760/Desiccant.Box.-.Box.XL.HatchBox.PLA.Standard.LW-0.4mm.LH-0.2mm.I-25.T-E.215C.B.65C.zip)

HellAholic, 2024-03-29:
Hey @gh13 ,
Could you report this issue in the plugin developer's repository? 
I see the issue when using the spoon anti-warping plugin but not when using tab+ anti-warping plugin.

closed 2024-03-29 by HellAholic

gh13, 2024-03-29:
> Hey @gh13 , Could you report this issue in the plugin developer's repository? I see the issue when using the spoon anti-warping plugin but not when using tab+ anti-warping plugin.

I have reported the issues with the dev. I used to use the tab+ plugin, but it requires some changes to the support settings & requires a more specific support/support brim configuration for them to generate correctly, especially things like reducing the support X-Y distance for them to connect to the model, which can mess up supports if you are using them elsewhere in the print. Whereas the Spoon Anti Warping is a bit more flexible with the support setup

gh13, 2024-06-04:
I've tested 5.7.2 with the STL file I'd previously had the issue with. 
I added the SpoonTabs, saved the project, restarted Cura & opened the project file again. The layout with 5.7.2 was preserved as it should be
I also tested the project file in 5.7.1 & 5.7 with the layout also being preserved there too. 
So I'm presuming that whatever fix was implemented either saves some additional information in the project file in order to preserve the layout or has updated some shared resources in my 5.7.x configuration folder, which is why it is now working for me at least with older versions too, which have previously been problematic

[image]

[Desiccant Box - Box [XL] [HatchBox PLA, LW-0.4mm, LH-0.2mm, I-25%, T-E.215C, B.65C].zip](https://github.com/user-attachments/files/15569151/Desiccant.Box.-.Box.XL.HatchBox.PLA.LW-0.4mm.LH-0.2mm.I-25.T-E.215C.B.65C.zip)