Ultimaker/Cura #18764: Slicing hangs

RyanelRichie, 2024-03-29:
### Cura Version

5.4.0

### Operating System

Mac  Ventura 13.6.6

### Printer

Ender 3 Pro

### Reproduction steps

Whenever slicing any STL, the slicing bar stops about half way and stays there. Have rebooted computer, app, different files, different settings. They all just hang. 
[image]

### Actual results

Nothing happens. Just hangs. 

### Expected results

Slicing would finish. 

### Add your .zip and screenshots here ⬇️

[CE3PRO_Faction_Fenris_Holder_Cover.3mf.zip](https://github.com/Ultimaker/Cura/files/14807209/CE3PRO_Faction_Fenris_Holder_Cover.3mf.zip)

HellAholic, 2024-03-29:
Can you try [the 5.7.0-beta](https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1)? We've fixed a lot of the slicing errors we saw in the 5.x-series in the upcoming 5.7.0 version.

RyanelRichie, 2024-03-29:
THAT WORKED!! Thank you so much for the quick response. Wasn't aware there was a newer beta. Thanks!

HellAholic, 2024-03-29:
Glad that helped.

closed 2024-03-29 by HellAholic