Ultimaker/Cura #18776: 'Group outer wall' true by default ender3v3se

Izzy5656, 2024-03-31:
### Cura Version

5.6.0

### Operating System

Windows

### Printer

Creality ender-3 v3 se

### Reproduction steps

Created a profile for the ender3 v3 se

### Actual results

After under extrusion issues I found 'group outer walls' under experimental was ticked by default 

### Expected results

This should be unticked for new users

I realised this setting probably doesn't trigger a retraction when travelling a long distance between different outer walls. This causes under extrusion starting a new outer wall where the z seams are far apart (like a hole far away from the outer z seam) 

Outer walls before inner walls hid this issue (but this isn't great for overhangs) 

First timer here, hopefully this is clear enough

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2024-04-01:
Thanks for the report.
This isn't a bug but rather the way your Creality printer definition is written.  The setting works on some printers and maybe not so well on others.  It is an item that can be taken care of in the printer definition file as an "override" or in a custom setting profile by de-selecting it.  I've had to do that with "Support Horizontal Expansion", "Skirt Height"  and a couple of others.
UltiMaker cannot be responsible for other manufacturers printer definitions.

I'll remove the bug label and go ahead and close this.  I hope you understand.

closed 2024-04-01 by GregValiant

Izzy5656, 2024-04-02:
Yeah, understand it's not really a bug unless not retracting between other outer walls is an issue with this feature when ticked (but I guess it is under experimental) . Took a while for me to figure it out and thought cura has control over the default settings and maybe this could be changed to save others having the same issue. Cheers👍

Izzy5656, 2024-04-09:
> Thanks for the report. This isn't a bug but rather the way your Creality printer definition is written. The setting works on some printers and maybe not so well on others. It is an item that can be [...]

Yeah, understand it's not really a bug unless not retracting between other outer walls is an issue with this feature when ticked (but I guess it is under experimental) . Took a while for me to figure it out and thought cura has control over the default settings and maybe this could be changed to save others having the same issue. Cheers👍

I've seen several new users have the same issue on Facebook groups

GregValiant, 2024-04-09:
The definition files come from members of the community.  If it's written OK codewise then it looks OK to the Cura team.  They don't have a way to tell if the settings are proper for a particular printer.
I look at the pull requests for some printers.  If the author is just adding a bunch of "personal preference" settings I try to dissuade them from forcing those settings onto all owners of whatever particular printer it is.  The settings we've talked about here are cases in point EXCEPT for Group Outer Walls which is a new setting.
The author of the printer definition should really be the one to do the background work on new settings to see if the setting is proper for the particular machine.
These are the last few lines of the Override section of my personal printer definition file (Ender 3 Pro).  I keep the file in the Definitions folder within my Configuration folder.

	"support_offset": { "default_value": 0.0 },
	"machine_endstop_positive_direction_z": { "default_value": false },
	"material_initial_print_temperature": { "enabled": "machine_extruder_count > 1" },
	"material_final_print_temperature": { "enabled": "machine_extruder_count > 1" },
	"group_outer_walls": { "value": false }
  }
}
The two temperature settings are only used by multi-extruder printers so I don't need to have them clutter things up.  The other two (Support Horizontal Expansion and Group Outer Walls) I don't use but are set by default.  Now they aren't.