Ultimaker/Cura #18783: Slicing failed Son't you have logs for this?

Dante-Evans, 2024-04-01:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

FLSUN V400

### Name abnormal settings

none

### Describe model location

Center of base

### Describe your model

This was sliced once but every other time it has failed

### Add your .zip here ⬇️

Your program only allows me to save the file as a *.3mf but you don't allow that kind of file to be uploaded seems like you aren't trying to help troubleshoot your own program to me. And of course since you won't believe me I am going to upload several screen shots showing that.
[image]
[FV_Chest_Back.zip](https://github.com/Ultimaker/Cura/files/14818656/FV_Chest_Back.zip)

Asterchades, 2024-04-01:
The inability to upload 3MF files directly isn't an Ultimaker limitation - it's a GitHub limitation, which is well and truly outside of Ultimaker's control. It's also an extremely easy limitation to bypass, given that every modern OS can easily compress a file and any OS with disc drive access can change the extension to one that GitHub will accept. Note that it is strictly a file name limitation as 3MF files are already ZIP files with a specific internal file structure, which would make them valid if any more thorough checks were actually performed.

To the issue at hand, try with the [5.7 beta release](https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1) currently available. This already includes a large number of fixes for slice engine crashes like this. It also includes better logging facilities - which you don't even need to interact with at all - for more accurately reporting where and how the engine crashed. Said facilities, while enabled by default, are optional and can be disabled.

GregValiant, 2024-04-01:
_"...And of course since you won't believe me..."_
I can't speak for the other people who do triage here, but for myself, that is 100% wrong.  I believe _EVERYTHING_ I read and see here and I believe that everyone who posts a bug report here does so in good faith.
**I know** there are bugs in the program, **you know** there are bugs in the program, and the team at **UltiMaker knows** there are bugs in the program.  That doesn't mean that every report involves a bug.  It's why things need to be investigated.
The reported problem could be a bug but my experience has shown that it could just as well be a model problem or a settings problem.

The "won't slice" bug started in 5.4 and is especially noticeable in Windows 11 and with "organic" shapes.  It has taken a couple of tries but the Cura team seems to have made great progress in 5.7.  I didn't have any trouble slicing your project in 5.6.0 but I'm on Win10 which hasn't had near the problems as Win11 has.

To help with the investigation you should take Asterchades advice and install the beta from [HERE](https://github.com/Ultimaker/Cura/releases) by scrolling down to "Assets" and picking the proper installation file.  5.7b may well resolve the problem OR it may take more digging to get to it.

I'll mark this as a duplicate but I'll leave it open for a bit to allow you to respond.  Let us know how it works out with 5.7b.

closed 2024-04-15 by github-actions[bot]