Ultimaker/Cura #18784: Initial Layer Temp Being Set at the Wrong Piont

dunedon, 2024-04-01:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Sovol SV03

### Reproduction steps

Set the Printing Temperature to 210
Set the Printing Temperature Initial Layer to 230
Slice File

### Actual results

When examining the G-Code (and on the printer) the initial temperature is set to 230 (Expected)
However, the temperature drops to 210 after the walls are printed and before the Initial Layer skin is printed (Not-Expected)
See the attached g-code file as well as below.  I would have expected this temp change AFTER the initial layer, not before.

`;TYPE:SKIN
G1 F2400 E150.79077
[2x10_NoIron_Bad.txt](https://github.com/Ultimaker/Cura/files/14823552/2x10_NoIron_Bad.txt)

G1 F900 X52.085 Y152.085 E163.05527
G1 X52.085 Y197.915 E165.34174
G1 X297.915 Y197.915 E177.60625
G1 X297.915 Y152.085 E179.89272
M205 X16 Y16
G0 F2160 X297.314 Y152.685
**M104 S210**
M205 X8 Y8`

### Expected results

I was expecting the GCODE to appear at layer 1 not where it did.

### Add your .zip and screenshots here ⬇️

[2x10_NoIron_Bad.txt](https://github.com/Ultimaker/Cura/files/14823562/2x10_NoIron_Bad.txt)
[2x10 Movement Tray.zip](https://github.com/Ultimaker/Cura/files/14823572/2x10.Movement.Tray.zip)

GregValiant, 2024-04-01:
Thanks for the report.
Where the "M104 S Print Temperature" line goes into the gcode is a function of the printer settings "Heat Up Speed" and "Cooldown Speed" both of which are in "  °C/second ".
I checked in 4.13.1 and it worked correctly, but in 5.6 and 5.7beta it appears to be broken.

@HellAholic and the Cura team will take a look.

HellAholic, 2024-04-01:
I can only reproduce if I use the custom profile "SV03 - Sunlu PLA...", using the standard quality profile the expected behavior and actual results match correctly.
Set the print temp for initial layer (S250)
[image]
Sets the target for printing temp before the first layer (S210)
[image]

dunedon, 2024-04-01:
I haven't done anything but set settings using the cura UI, so not sure why that profile is messing up?

closed 2024-04-01 by GregValiant

dunedon, 2024-04-01:
I did a search on trying to find where I could set the Heat Up and Cool Down values (and I can't figure out where to set those anywhere) ... but I did notice that very similar problems have been previously reported, and haven't been fixed as of the end of last year, so I don't think it's just my profile causing the problem.

Where can I change the Heat Up and Cool Down values? if I can do that I can at least trick the system into working for me as I want it.