Ultimaker/Cura #18795: nothing to support

illusionmanager, 2024-04-02:
### Cura Version

5.6.0

### Operating System

windows10

### Printer

snapmaker with dual head

### Reproduction steps

load model, enable support, slice.

### Actual results

In the sharp corners cura generates 1 layer of support, even though nothing is above the support. See attached image It get even worse when increasing the value of "support horizontal expansion" from 0.4 to 2mm, in which case support is al around the object.

### Expected results

none of it should be generated

### Add your .zip and screenshots here ⬇️

[moon-rise-pointer.zip](https://github.com/Ultimaker/Cura/files/14833717/moon-rise-pointer.zip)
[image]

HellAholic, 2024-04-02:
Hey @illusionmanager,
This is an issue with your printer definition or profile. Since neither of which is included within Cura, we cannot help you further with this. The image below shows a Snapmaker 2 A350 Dual.
There is a warning when importing your project file that indicates a chance for some or more settings could have been corrupted.
Regardless of issue mentioned above, we improved few functionalities with regards to the support (normal and tree), they're available in [5.7 Beta](https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1) and will be part of the stable release coming soon. You can give that one a try as well.

[image]

I am closing this issue since we're not able to reproduce it and the signs points us towards changes in your settings.

closed 2024-04-02 by HellAholic

illusionmanager, 2024-04-02:
i found the problem. It's not a machine profile. It has to do with the cura setting "initial layer horizontal expansion" which I had set to -0.2mm. Apparently cura things it needs to build support for that. So I still think it is a bug. You don't generate support for thing as small as half the nozzle width.

HellAholic, 2024-04-02:
There is a setting "Minimum Support Area" which you also need to adjust when changing the initial layer horizontal expansion in combination with support, if you do not which to see such artifacts.
for -0.2mm the value of 3.2 squared mm will remove the artifacts you see and gives you the support for 0.16mm layer height.

[image]

illusionmanager, 2024-04-02:
Yes, that worked. Thank you. ((( but still it is a bit silly to generate support at a position where there is actually nothing above the support)))