Ultimaker/Cura #18799: Paint on Supports

HermanGaviriaMarin, 2024-04-02:
### Is your feature request related to a problem?

Support Blockers are the absolutely tedious and non intuitive at all. They require so much more work to get right as we have to scale them, then rotate, or move in 1 axis at a time. This adds up specially with complex models which require unique supports and support blocking. 

### Describe the solution you'd like

Paint on supports and paint on support blockers

### Describe alternatives you've considered

Using Prusa slicer, but who wants to switch ?? Not me 

### Affected users and/or printers

Me myself and I and the entirety of the cura userbase

HermanGaviriaMarin, 2024-04-02:
I know this topic has been discussed in the past already. But I think its important bring back up because we have been waiting a long time for this. Several months ago if not already a year and a few months there was a ton of talk about this being a feature which was highly requested.

rburema, 2024-04-03:
We realize this is an important, often requested feature. We've actively investigated it a couple of times. While I can't give a timeline, we're also convinced that it should be added to Cura at some point.

rburema, 2024-04-03:
A little update: Can report that one of the tickets in our current sprint is related to seeing if an old project we had in the works for this can be revived, or if the code-rot is too bad by now.

HermanGaviriaMarin, 2025-12-09:
"Paint on seam and material" have been added but I haven't seen the paint on supports anywhere, I've scoured the settings visibility to look for say setting but no luck. If you're referring to the seam and material brush yea its there. But It might be a early to say this specific feature is there until, you know, its there. 

I'm ecstatic to see a similar feature has made its way into Cura though, cheers to that team.

rburema, 2026-10-08:
Support Painting is definitely going to be in _**5.14**_ -- look for the upcoming **Beta** of that version that should be **out very soon**!

P.S. If you really can't wait, there's a slightly scuffed version of support painting in the Alpha of 5.14 [here](https://github.com/Ultimaker/Cura/releases/tag/5.14.0-alpha.0) -- but honestly, the Beta release is planned so close that I don't think it's worth installing that one.

closed 2026-10-08 by rburema