Ultimaker/Cura #18801: Error al trasladas la figura en vista previa

Francisco-Javier-Carreon-Duran, 2024-04-03:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

kingroon kps3

### Name abnormal settings

no

### Describe model location

no

### Describe your model

si

### Add your .zip here ⬇️

[image]

HellAholic, 2024-04-03:
Hey @Francisco-Javier-Carreon-Duran , 
Can you provide some information here? We can't help you with your issue at the moment.
Please use the fields to describe your issue rather than answering Yes/No to the example questions there.
Add steps to reproduce or a model. And lastly let us know what is the expected behavior and what is the actual results that is causing the issue for you.

rburema, 2024-04-03:
Could you try using the new 5.7.0? Does the issue still happen in that version. As I recall, there might have been some changes to how we handle the tool-affordances since 5.4.0

closed 2024-04-17 by github-actions[bot]