Ultimaker/Cura #18802: slicing failed due to some error

Yellouublack, 2024-04-03:
### Cura Version

5.7.0

### Operating System

windows 11

### Printer

anycubic kobra neo

### Describe your model

qr code which was downloaded from internet was not to slice

### Add your .zip here ⬇️

this error shows while slicing 
"Oops! We encountered an unexpected error during your slicing process. Rest assured, we've automatically received the crash logs for analysis, if you have not disabled data sharing in your preferences. To assist us further, consider sharing your project details on our issue tracker."
[image]

HellAholic, 2024-04-03:
Hey @Yellouublack ,
Without a project file we cannot help you further. As the message mentions, we have received a report on our analysis tool (if you're connected to internet on your machine and allow network access). If you wish to help us further or enable us to track your issue better, a project file is a must.

closed 2024-04-17 by github-actions[bot]

trisdiastato, 2024-06-04:
same problem guys (Oops! We encountered an unexpected error during your slicing process. Rest assured, we've automatically received the crash logs for analysis, if you have not disabled data sharing in your preferences. To assist us further, consider sharing your project details on our issue tracker.)   I use 3 years now Cura slicer and with last update on Mac 5.7.1. I have problem on slice