Ultimaker/Cura #18803: Can't run Cura with Nvidia proprietary drivers in Linux -  Issue with a signal handler in one of the QML files. versionFunctions: Not supported on OpenGL ES

erickwill, 2024-04-03:
### Cura Version

5.7.0 and 5.6.0

### Operating System

Arch Linux running GTX1650 through proprietary drivers and nvidia-prime

### Printer

Tronxy X5SA (Don't related though)

### Reproduction steps

Trying to running Cura through Nvidia card using Nvidia's proprietary drivers through both nvidia-prime and bumblebee.

Seems like.. This error is to be related to a bug or issue within the Cura application itself, specifically with its QML files and rendering/context setup.

Here's a breakdown of the different error messages:

QEGLPlatformContext: Failed to create context: 3009
This suggests an issue with creating the OpenGL context required for rendering the UI.
qt.qml.typeresolution.cycle: Cyclic dependency detected between qml files
This indicates a cyclic import dependency issue between the QML files used by Cura.
Binding loop detected for property "height/width"
There seems to be a binding loop issue in one of Cura's QML components (PowerCostWidget).
Parameter "index" is not declared.
An error related to a signal handler in one of the QML files.
versionFunctions: Not supported on OpenGL ES
This could be a warning related to the OpenGL ES implementation on your system.
These errors collectively point to some compatibility issues or bugs within the Cura application itself when running Linux along with Nvidia proprietary drivers and interface.

### Actual results

QEGLPlatformContext: Failed to create context: 3009 qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_UltiMa9CdB41/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMa9CdB41/share/cura/resources/qml/Actions.qml" file:///home/erick/.local/share/cura/5.7/plugins/PowerCost/PowerCost/PowerCostWidget.qml:13:2: QML PowerCostWidget: Binding loop detected for property "height" file:///home/erick/.local/share/cura/5.7/plugins/PowerCost/PowerCost/PowerCostWidget.qml:13:2: QML PowerCostWidget: Binding loop detected for property "height" file:///home/erick/.local/share/cura/5.7/plugins/PowerCost/PowerCost/PowerCostWidget.qml:12:2: QML PowerCostWidget: Binding loop detected for property "width" qt.qml.context: file:///tmp/.mount_UltiMa9CdB41/share/cura/resources/qml/Settings/SettingView.qml:427:16 Parameter "index" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead. versionFunctions: Not supported on OpenGL ES

### Expected results

To work as other apps do.

### Add your .zip and screenshots here ⬇️

Don't need it
[image]

erickwill, 2024-04-03:
> In addition to what @smartavionics says, is the the package we offer for download, or Arch' own package?
> 
> In any case, this is a new one, it really _shouldn't_ be getting the ES version of OpenGL, as we set `Qt.ApplicationAttribute.AA_UseDesktopOpenGL`, which is supposed to force it to the desktop implementations.
> 
> (As an aside, a lot of the binding loops come from the 'PowerCostWidget' plugin you seem to have installed.)

Sorry, but I'm trying to report an issue while using Nvidia, not bashing anyone work here. Yes, I'm using the package provided by you guys and no, the "PowerCostWidget" that I have installed is not the culprit for the issue I'm facing, 64GB of ram, I think is sufficient to get both running. 
Curiously, I can run many other apps that uses OpenGL without a single problem. Maybe is a widget issue.

erickwill, 2024-04-03:
> Little bit of info. Cura works with the desktop version of OpenGL and not the ES version. I have ported Cura to OpenGL ES as that is supported on the Raspberry Pi. I would have thought it unlikely that the Cura devs will make it compatible with ES. Why doesn't your system support desktop OpenGL?

It does, in fact, for this reason I'm reporting this issue.

rburema, 2024-04-03:
> Sorry, but I'm trying to report an issue while using Nvidia, not bashing anyone work here. 

I get that, just covering all bases here. I wasn't trying to blame you, just a brain dump of how the issue appeared to me. I'm also not trying to deny that the issue might be within Cura (or, in worst case, since we can't fix that, QML, which seems to happen more frequently lately...)

> and no, the "PowerCostWidget" that I have installed is not the culprit for the issue I'm facing

I wasn't really implying that. Just that those parts of the error message are probably not related.

> Curiously, I can run many other apps that uses OpenGL without a single problem.

That might help us pin it down, depending if those apps are fine with web-OpenGL or not.

MariMakes, 2024-04-05:
Hi @erickwill,

We discussed it with the team and we believe we will not work on this anytime soon and will therefore defer this issue so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker. Does the fork from @smartavionics get you unstuck? 

We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.  So, thanks again, and please let us know if you run into any other issues 💪 

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

wosym, 2024-06-14:
@erickwill did you find a solution for your problem? I am encountering the exact same issue. 

I'm puzzled why it's loading the ES version of OpenGL while I'm obviously on a desktop?

elprocrastinador, 2024-09-09:
This guy can run cura as root. Not me.

https://forums.gentoo.org/viewtopic-t-1170533.html?sid=b2505237b861438b9375654258454e57

Version override doesn't work
```
# echo "MESA_GL_VERSION_OVERRIDE=4.6" >> /etc/environment
# echo "MESA_GLSL_VERSION_OVERRIDE=460" >> /etc/environment
```

ErikMichelson, 2024-09-09:
Using NVIDIA drivers 555.58.02 under Wayland but with x11-fallback (setting `QT_QPA_PLATFORM=xcb`) the print bed and models are visible and the app is performing right.
However, not having to fall back to x11 is of course still preferred.

elprocrastinador, 2024-09-09:
Sorry but I don't know how to implement that setting.
Can you explain it?
Thanks

bougar, 2024-09-11:
Hello!!

I am having the same problem in Fedora 40. Running with QT_QPA_PLATFORM=xcb shows the bed but nothing when loading models.

bougar, 2024-09-11:
> Sorry but I don't know how to implement that setting. Can you explain it? Thanks
> 
> edit: OK, I know. $$ QT_QPA_PLATFORM=xcb ./pathto/UltiMaker-[Cura-5](https://ultimaker.atlassian.net/browse/CURA-5).8.1-linux-X64.AppImage
> 
> Now machine bed is shown but does nothing when loading models.

This is finally working for me:

* Run Cura setting `QT_QPA_PLATFORM` variable to `xcb`:
`QT_QPA_PLATFORM=xcb ./pathto/UltiMaker-Cura-5.8.1-linux-X64.AppImage`
* On top-right click Marketplace:
* Click on the gear icon (Manage Packages)
* Search `USB Printing` and disable it.
* Restart Cura and try loading a model again.
* That is working for me.

Good Luck!

m-vanson, 2024-11-12:
I was having the same issues.

I'm on OpenSuse Tumbleweed with an NVIDIA card. What worked for me is setting the permissions in Cura to force it to use X11 instead of Wayland:
[image]
Click on the Advanced permissions:
[image]
Disable Wayland (I'm not sure if you need the enable "Fallback to X11")