Ultimaker/Cura #18807: Cura Crashes When Uploading To Octoprint

NokEternal, 2024-04-03:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Creality Ender 3 S1 Pro

### Reproduction steps

Sliced a file, tried to upload it to Octoprint

### Actual results

Cura Shuts down a few seconds after rendering file for upload to 
Octoprint.

### Expected results

File Should load successfully and prompt a link to Octoprint.

### Add your .zip and screenshots here ⬇️

I have no screenshots to use because cura just closes.

wawanbreton, 2024-04-04:
Hi @NokEternal , thanks for the report !
This issue is quite annoying, but I fail to see a reason why Cura would crash right now.
Did you run previous versions of Cura before ?
It would greatly help us if you can [provide your log files](https://support.makerbot.com/s/article/1667337559413) so that we can see what is happening.
And by the way, Octoprint access is a Cura plugin that is not developped by us, so you may get more relevant help [on the plugin page](https://github.com/fieldOfView/Cura-OctoPrintPlugin).

NokEternal, 2024-04-05:
> Hi @NokEternal , thanks for the report ! This issue is quite annoying, but I fail to see a reason why Cura would crash right now. Did you run previous versions of Cura before ? It would greatly help [...]

I had to reread your comment a few times to get the gist of what you were asking for lol, anyhow I did look through all my folders and whilst I found how to get into said folders, I don't actually see any cura.log file to share with you, i see all other folders but no actual log, i will take a screenshot of this and maybe i'm just confused? 
[image]

wawanbreton, 2024-04-08:
> I don't actually see any cura.log file to share with you

Well I do see it on your screenshot :smiley: it is actually the "cura" file referenced as being a "Text Document". By default, Windows hides the files extension of common types (which is a nightmare), so in this case the file is indeed called "cura.log" by it will only display "cura".

NokEternal, 2024-04-08:
> > I don't actually see any cura.log file to share with you
> 
> Well I do see it on your screenshot 😃 it is actually the "cura" file referenced as being a "Text Document". By default, Windows hides the files extension of common types (which is a nightmare), so in this case the file is indeed called "cura.log" by it will only display "cura".

Okay so with all that said and me finding the log files, how do i go about uploading them to you? Sorry for any inconvenience I am only about a year deep in to 3d Printing as well as working with the Cura Slicer, so some things take a minute for me to catch up lol, anyhow i do have said log file, but it wont let me just upload it through here, so with that being said any more help on this situation would be greatly appreciated, as I love Cura for my slicer but this issue is causing me quite the delay on getting things back and up and running. Thank you for your time.

NokEternal, 2024-04-08:
[cura.log](https://github.com/Ultimaker/Cura/files/14907660/cura.log)

NokEternal, 2024-04-10:
Just checked again and the problem is still persisting, It's with any file and like I said initially, Cura doesn't respond with any known error code or failure, it simply just closes out as soon as I click the "Print with Octoprint" button,.. Which is odd as I do believe 5.7 is the newest version out for cura, this didn't happen to me on 5.6 before the initial switch over.. 
[image]

wawanbreton, 2024-04-17:
Hi, and sorry for the delay. I took a deep look at your log file, but couldn't find anything relevant. I do see the application suddenly stopping to log, but no indication of why it would have crashed. I see you have already opened a new issue on the octoprint plugin page, hopefully they can help you more.

NokEternal, 2024-04-18:
Hopefully in the next cura update this issue can be addressed.. Whilst annoying there is actually a work around that i've had to resort too, in order to keep printing with Cura, If anyone is having this same issue, the work around is quite simple. Slice said file in preferred settings and then "export g code" and then with octoprint upload file manually/locally and it seems to work fine then. Anyways i'll keep checking in occasionally with any if at all updates on this matter. thank you for those who have stayed persistent with this issue, as it really is rather annoying.

wawanbreton, 2024-04-19:
I'm afraid we (as Cura team) won't address this issue because this is related to the plugin. I hope @fieldOfView can help you on this. But I'm a bit puzzled because you seem to be using the plugin in the most standard way, and you have a very repeatable crash that is completely blocking. So we should see more people having the same issue, but maybe this is related to something in your specific environment. Have you tried the classic method of uninstall Cura - clear all configuration - reinstall Cura ?

NokEternal, 2024-04-19:
> I'm afraid we (as Cura team) won't address this issue because this is related to the plugin. I hope @fieldOfView can help you on this. But I'm a bit puzzled because you seem to be using the plugin [...]

I actually have not, whilst I am familiar with uninstalling a program, I haven't done so as of yet but am down to give it a shot, just so i'm clear on this process, how would I go about clearing the entirety of the cura configuration contents to help assist with said dilemma??? I  just want to make sure i'm thorough on these next few steps, so I can in fact see if it is something I'm perhaps doing wrong? anymore help on this situation is as always very appreciated. I just want to get back to printing without having to do any of the above said work around's and or possible bug that I am currently on hand with, Thank you so much again on this matter, I very much do appreciate any and all effort towards this issue..

Aynshe, 2024-05-05:
I'm seeing the same crash issue when sending to OctoPrint since 5.7.0.

v5.6.0 works fine when sending the file.

FunVeee, 2024-05-05:
I'm having the exact same problem since 5.7.0... It is NOT the plugin as it used to work fine.

NokEternal, 2024-05-06:
Whilst it is nice to see others and me not being the only one in the entire world having this problem, it is also quite rather frustrating, The only work around I have found using Cura 5.7 was to export the g.code file after being sliced and then uploading it locally via the octoprint homescreen. It is rather annoying though not having that instant gratification, I do indeed hope the cura team can address this issue, as I am not the only one...

wawanbreton, 2024-05-06:
> I am floored that they are saying it MUST be YOUR environment because you are the only person to take the time to submit an issue... LAME AF!! It is Cura 5.7, not the environment.

Well now we have two users having the same issue, we may consider that this is not the environment indeed, but in the first place we have to assume it, because it is a major root cause of issues.

> It is rather annoying though not having that instant gratification, I do indeed hope the cura team can address this issue, as I am not the only one...

I agree that this is a very annoying issue. I will bring it for discussion with the team and see if we can spend some time on it, to at least try and find what is causing it.

> I'm having the exact same problem since 5.7.0... It is NOT the plugin as it used to work fine.

Somehow the plugin may need an update because of the changes introduced in 5.7. So you are right, the plugin itself is not faulty indeed, but it is like "no more compatible" with the new version. We have a lot mechanisms in place to avoid those situations, but it is quite impossible to completely erase them.

wawanbreton, 2024-05-08:
FYI, we will be taking a look at this in the next sprint.
For internal reference: CURA-11901

wawanbreton, 2024-05-14:
Hi,
I was able to do some tests this morning, but unfortunately I couldn't reproduce the crash :confused: 
Here are the environments I tested:
* Ubuntu 22.04.4 + Cura from source
* Ubuntu 22.04.4 + Cura 5.7.0 AppImage
* Windows 11 (VM) + Cura 5.7.1

I really can't fix this if I can't reproduce it, so can everyone encoutering this bug please give me as much information as possible:
* OS version
* Cura version and how it was installed (exe, msi, other)
* OctoPrint plugin version and how it was installed (manually, from marketplace)
* OctoPrint server version
* Printer
* Ideally, a 3MF project file with the printer setup
* A brief step-by-step explanation of how you use Cura until it crashes (ideally a desktop record)

@NokEternal The Windows setup I tested is very close to yours, so I am really confused. I will try to see if I can test on a Native Windows 11 with Cura 5.7.0 to be even closer.

wawanbreton, 2024-05-14:
@NokEternal I also see on your screenshot that you have a few UI plugins enabled. Can you try to disable as many plugins as possible and see if it still crashes ?

NokEternal, 2024-05-14:
Windows 11 is my OS, 
Cura Version 5.7-Installed from ultimaker.com and I used the exe. file
Octoprint 3.7.3 from the marketplace within cura itself.
Octoprint 1.10.1 (Newest Server Update as of 05/14/24)
Ender 3 S1-Pro is my printer that I have been using.

As far as screen record goes I don't have that capability as of now on said computer but I could probably manage sometime later this evening trying to screen cap the bug in question that keeps happening, I will be uploading that within the next few hours if not sooner, and as far as step by step details go, I simply just slice any file and try to within the cura field, click onto the upload to octoprint button, and within a few seconds after that, Cura ends up closing out, with no trace of any error report.. I will conduct some tests here shortly and as well as try to disable as many UI plugins as well and see if that helps.

NokEternal, 2024-05-14:
UPDATE: I'm not quite sure what could be causing it, maybe possibly a UI Setting within Cura but I have since tried uninstalling as many plug ins, and managed to get it to work like normal, I just only wish I could specify what plugin could possibly be causing this issue, but it has since been resolved at least for me, I will conduct furthermore tests and see if I can possibly pinpoint what Plugin(s) could be causing the initial issue..

FunVeee, 2024-05-14:
In case it helps, here are my settings that are still causing it to crash.
Windows 11
Cura 5.7.0
Octo4a from GitHub (Running on an Android Samsung S8)
Elegoo Neptune 3 Plus

wawanbreton, 2024-05-15:
@NokEternal Thanks for this amazing investigation work :smiley: Now I will try to install the ELEGOO NEPTUNE THUMBNAILS plugin and see if I can reproduce the bug, but I'm pretty positive !

@FunVeee Do you happen to have this plugin installed ? I see you do have an Elegoo printer, so that would make sense..

FunVeee, 2024-05-16:
You are correct, I do have the Neptune thumbnails installed.  Great detective work!!

NokEternal, 2024-05-16:
Glad we could sort things out. Sorry for the assumption that this  was either Cura and or Octoprint. I had just never ran into any issue like this before and just had assumed, Either way IDK if the Elegoo plugin needs an update possibly or just bugged out, hopefully this can be resolved for people in the near future if need be, Thanks again for the assistance and happy printing everyone!

wawanbreton, 2024-05-17:
> Sorry for the assumption that this was either Cura and or Octoprint

Well at some point you have to assume something :smiley: 

So I made some investigation yesterday, and it turns out that it is not really the Elegoo plugin itself, but a Cura method that is called by it. Now the issue is, the crash occurs in some deep PyQt/OpenGL interaction, and this is something that is really tricky to fix :grimacing: We did upgrade to a new version of PyQt for Cura 5.7, which is supposed to make it globally more stable, but since then we have encountered this kind of bugs a few times, without being able to really fix them.

For the moment I would just advise to keep the Elegoo plugin disabled, which is **I think** a bit more expendable than the Octoprint one for the everyday use.

lint-os, 2024-07-11:
I think I have a method to reproduce the issue:
Install Cura -> set profile to Creality Ender 3 s1 pro -> add octoprint plugin -> connect to server and configure octoprint plugin like picture -> add gcode thumbnailer post processing script (the builtin one) -> slice any file and attempt to upload to octoprint -> hang, then crash

To stop it from crashing either check 'Start print job after uploading' and uncheck 'Confirm print job options before sending' or disable gcode thumbnailing.

it seems that whenever the confirm settings window appears when gcode thumbnailing is enabled, cura crashes.
[image]

conan104, 2024-09-27:
hello, unfortunately from versions after 5.6.0 I have encountered a Cura crash when I try to print using DUET RepRap Integration plugin (ver: 1.2.11).
Cura versions tested 5.7.0 ....... 5.8.1
Ubuntu 24.04.01 LTS.
I don't think the plugins are the problem (all removed, except DUET). The only solution I found to overcome the problem is to save the project before printing with Duet RepRap.
This procedure is not very convenient, are there any suggestions?